tube0013/tube_gateways #86: Failure to flash 20220928 zstack beta code using bsl

davidcoulson, 2022-10-01:
Attempting to try the new beta zstack code on my tube-zb-gw-cc2652p2-v2 - was previously running 20220219

david@djc-ubuntu1:~/cc2538-bsl$ ./cc2538-bsl.py -p socket://10.2.3.70:6638  -evw ../CC1352P2_CC2652P_launchpad_coordinator_20220928.hex
Opening port socket://10.2.3.70:6638, baud 500000
Reading data from ../CC1352P2_CC2652P_launchpad_coordinator_20220928.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:25:8F:28:85
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'

Was able to reboot it and flash it back to 20220219 - Not sure if issue with zstack or the coordinator. Has anyone else tried 20220928 on this device yet?

Will try to flash via serial later once I don't have to worry about my kids complaining the lights don't work :)

tube0013, 2022-10-01:
I recommend doing it over serial it will go much faster and less chances for disconnects.

Network flash should work but it appears to be less reliable and I'm actively looking into it.

davidcoulson, 2022-10-01:
Finally got 20220928 working - Ended up flashing via a system on the same VLAN which completed successfully multiple times. 

As usual, it's the network's fault :)

BTW I just ordered the new 2022 PoE CC2652P coordinator - Should be able to use the zigpy tools to backup the nvram off this one and restore it to the new one, right?

tube0013, 2022-10-01:
z2m should just migrate at startup if they see a blank radio.

Zha you can go through the new migrate flow