tube0013/tube_gateways #90: zigbee2mqtt not working

scfigg, 2022-10-07:
I migrated over to the 26252P2 from a usb stick that ended up dying. I deleted the zigbee2mqtt addon in my HA instance since it wasn't working off the bat. I have regular zigbee working - but can't get zigbee2mqtt working. I have a fresh install of the z2m addon with basic configuration - but after a couple of minutes, it stops with this error:

[16:10:23] INFO: Preparing to start...
[16:10:23] INFO: Socat not enabled
[16:10:24] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-10-07 16:10:25: Logging to console and directory: '/config/zigbee2mqtt/log/2022-10-07.16-10-25' filename: log.txt
Zigbee2MQTT:info  2022-10-07 16:10:25: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-10-07 16:10:25: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-10-07 16:10:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-07 16:10:45: Failed to start zigbee
Zigbee2MQTT:error 2022-10-07 16:10:45: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-07 16:10:45: Exiting...
Zigbee2MQTT:error 2022-10-07 16:10:45: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)

scfigg, 2022-10-07:
[image]

tube0013, 2022-10-07:
Make sure zha is completely disabled and not trying to /connected to the device while z2m starts.

You can also watch the web console on the device (go to the ip in browser) to see if anything is connecting to serial.

tube0013, 2022-10-07:
Only one host application (zha or z2m) can use a coordinator at a time. The coordinator only allows one serial connection at a time for this reason.

scfigg, 2022-10-10:
thank you - I did this, and got it working.

Stupid question again now - are zigbee routers not supported or work as traditional routers? I have a few bulbs and cc2530 diy routers paired with the coordinator, but everything maps to the coordinator it looks like - nothing is bouncing off each other or the closest router. Is this normal?  

[image]

tube0013, 2022-10-10:
It should change over time. If not you can re pair a device to a specific router but only enabling joins through the desired router. Device may move parents though after this depending on the network and devices.