tube0013/tube_gateways #95: Z2M not started with tube_zb_gw_cc2652p2_poe

idMasjid, 2022-10-14:
Hello,
I have a tube_zb_gw_cc2652p2_poe I installed Z2M on portainer but it won't start.
Z2M logs:
`Zigbee2MQTT:info  2022-10-14 23:41:57: Logging to console and directory: '/app/data/log/2022-10-14.23-41-57' filename: log.txt
Zigbee2MQTT:info  2022-10-14 23:41:57: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-10-14 23:41:58: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-10-14 23:42:17: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-14 23:42:17: Failed to start zigbee
Zigbee2MQTT:error 2022-10-14 23:42:17: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-14 23:42:17: Exiting...
Zigbee2MQTT:error 2022-10-14 23:42:17: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-10-14 23:42:18: Logging to console and directory: '/app/data/log/2022-10-14.23-42-18' filename: log.txt
Zigbee2MQTT:info  2022-10-14 23:42:18: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-10-14 23:42:19: Starting zigbee-herdsman (0.14.62)
`
log for my gateway
`[D][streamserver:085]: New client connected from 192.168.88.218
[D][streamserver:050]: Client 192.168.88.218 disconnected
[D][streamserver:085]: New client connected from 192.168.88.218
[D][streamserver:050]: Client 192.168.88.218 disconnected`

and my configuration.yaml

`# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: tcp://192.168.88.239:6638
`
thanks in advance

tube0013, 2022-10-14:
Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms) at ZStackAdapter.start 

Please make sure zha is disabled and not trying to connect at the same time. 

If not. Please try reflashing the Esp32-PoE module (best if done over serial)

Round model esphome firmware binary here: https://github.com/tube0013/tube_gateways/tree/main/models/retired/tubeszb-cc2652-poe/firmware/esphome

You can also try the beta esphome image based on latest esphome version. 

To flash over serial connect device to computer with micro usb. Be sure **NOT** to connect PoE at the same time! You may have to remove the device from the enclosure to access the port.

Flashing instructions are [here](https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-cc2652-poe-2022/readme.md) esp32 flashing section - but be sure to grab the binary for the model I linked above if the round PoE model. 

Can answer any questions you may have.

idMasjid, 2022-10-18:
Hello,
I just flashed my tube_zb_gw_cc2652p2_poe and I restarted my Z2M and I get another error.
Here are the logs:

Z2M logs:
```
Zigbee2MQTT:info  2022-10-19 00:30:33: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-10-19 00:31:40: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-19 00:31:40: Failed to start zigbee
Zigbee2MQTT:error 2022-10-19 00:31:40: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-19 00:31:40: Exiting...
Zigbee2MQTT:error 2022-10-19 00:31:40: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-10-19 00:31:42: Logging to console and directory: '/app/data/log/2022-10-19.00-31-41' filename: log.txt
Zigbee2MQTT:info  2022-10-19 00:31:42: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-10-19 00:31:42: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-10-19 00:32:49: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-19 00:32:49: Failed to start zigbee
Zigbee2MQTT:error 2022-10-19 00:32:49: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-19 00:32:49: Exiting...
Zigbee2MQTT:error 2022-10-19 00:32:49: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-10-19 00:32:51: Logging to console and directory: '/app/data/log/2022-10-19.00-32-51' filename: log.txt
Zigbee2MQTT:info  2022-10-19 00:32:51: Starting Zigbee2MQTT version 1.28.0 (commit #03ba647)
Zigbee2MQTT:info  2022-10-19 00:32:51: Starting zigbee-herdsman (0.14.62)
```

log for my gateway```
```

[D][serial_server:103]: New client connected from 192.168.88.218
[D][binary_sensor:036]: 'Serial Connected': Sending state ON
[D][serial_server:068]: Client 192.168.88.218 disconnected
[D][binary_sensor:036]: 'Serial Connected': Sending state OFF
[D][serial_server:103]: New client connected from 192.168.88.218
[D][binary_sensor:036]: 'Serial Connected': Sending state ON
[D][serial_server:068]: Client 192.168.88.218 disconnected
[D][binary_sensor:036]: 'Serial Connected': Sending state OFF
[D][serial_server:103]: New client connected from 192.168.88.218
[D][binary_sensor:036]: 'Serial Connected': Sending state ON
[D][serial_server:068]: Client 192.168.88.218 disconnected
[D][binary_sensor:036]: 'Serial Connected': Sending state OFF
[D][serial_server:103]: New client connected from 192.168.88.218
[D][binary_sensor:036]: 'Serial Connected': Sending state ON
[D][serial_server:068]: Client 192.168.88.218 disconnected
[D][binary_sensor:036]: 'Serial Connected': Sending state OFF
[D][serial_server:103]: New client connected from 192.168.88.218
[D][binary_sensor:036]: 'Serial Connected': Sending state ON
[D][serial_server:068]: Client 192.168.88.218 disconnected
[D][binary_sensor:036]: 'Serial Connected': Sending state OFF
[D][serial_server:103]: New client connected from 192.168.88.218
[D][binary_sensor:036]: 'Serial Connected': Sending state ON
```

and my configuration.yaml

```

# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: tcp://192.168.88.231:6638 ```
```

Thanks in advance,

tube0013, 2022-10-18:
Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

Try unscrewing the antenna during startup. Or unplug any near by routers temporarily.

tube0013, 2022-10-18:
Also see:

https://github.com/Koenkk/zigbee2mqtt/issues/10858