tube0013/tube_gateways #97: Router not connecting to devices

chicknlil, 2022-10-22:
I'm pretty sure I'm doing something wrong (in fact, I'm not entirely sure my ConBee II is set up correctly) but while I can get the coordinator to see your router (a floor below it, as it were), trying to add my only two current zigbee devices (Aqara Vibration Sensor x2) to it is failing miserably. I originally had them paired to the Zigbee network connector, but adding the router didn't seem to do anything - even though it was significantly closer to both of them, they didn't reconnect to it, instead, which from reading reviews, seems is what is the expected behavior. Don't know what I'm doing on, or if I've just gotten way over my head.

Any direction you could provide would be greatly appreciated. I will happily try to provide any information I can.

tube0013, 2022-10-23:
Hi, what are you using with the conbee? Zha/z2m/deconz?

Are yoga able to specifically try adding the sensors via the router?

chicknlil, 2022-10-23:
> Hi, what are you using with the conbee? Zha/z2m/deconz?
> 
> Are yoga able to specifically try adding the sensors via the router?

I have tried all of the above.  Deconz didn't let me add anything, period.  ZHA lets me add devices, but only from the controller, not from the router.  And this shows the visualization.
[image]

Feel free to poke me to provide more info.  I don't know what else you might be looking for.

tube0013, 2022-10-23:
You've gone to the router device page and hit the 3 dots and selected add via this device? Then attempted to repair the sensor?

chicknlil, 2022-10-23:
> You've gone to the router device page and hit the 3 dots and selected add via this device? Then attempted to repair the sensor?

I have, but let me try again.  I need to go to the basement anyways to poke at the Philips light down there.

chicknlil, 2022-10-23:
[image]

Does that help at all?

chicknlil, 2022-10-24:
I've added the second device I purchased from you to the basement. Not sure why, but it went unavailable 16 hours ago.  And yes, I did first try to connect it to the other router, but it wouldn't find it, and I ended up connecting it to the controller.

[image]

Frustrating me, though I'm sure it's something I'm doing wrong rather than your product.

tube0013, 2022-10-24:
Can you try and run an energy scan via this commandline tool:

https://github.com/zigpy/zigpy-cli

you'll need to disable ZHA

the install guide here is good for get it running in a terminal either on HAOS or another machine:

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

just move to the zigpy-cli install command once you have terminal in HAOS or in a venv.

then the energy-scan is: https://github.com/zigpy/zigpy-cli#performing-an-energy-scan

```
zigpy radio deconz /dev/ttyUSB0 energy-scan
```
(Change the port to match your system)

chicknlil, 2022-10-24:
Clearly, I'm an idiot, because I'm fucking even this up...

I installed the higher level Terminal & SSH. Put in my username and password in the config. Started the UI, and put in the install line, and get the following:

[image]

What am I doing wrong?

chicknlil, 2022-10-24:
Okay, I'm clearly an idiot, because I'm fucking even THIS up...

I installed the elevated Terminal and SSH addon, put in my username/password into the configuration. Checked the log to make sure everything looked good, than started the UI.  Pasted the command from the first link you provided.  And I get this...
[image]
What am I doing wrong THIS time? Am I just too stupid for HA?

tube0013, 2022-10-24:
First up don't worry, you aren't an idiot, I did the same thing initially, the copy line includes the $ at the beginning. 

but here are the steps cleaned up

Disable ZHA:

[image]

double check it's this addon:

[image]

Which I think it is.

disable protection mode and restart the addon.

[image]

run this to get inside the HA docker container:

```
docker exec -it homeassistant bash
```

then run the pip command:

```
pip install git+https://github.com/zigpy/zigpy-cli.git
```

Then you should be able to run the energy scan command:

```
zigpy radio deconz /dev/ttyUSB0 energy-scan
```

(changing the port - you shoud be able to find it through tab completion - start typing /dev/serial/by-id/ then hit tab until the right port is selected)

tube0013, 2022-10-24:
(proof I did the same thing)

[image]

chicknlil, 2022-10-24:
I ran it and it's still running, and I'm not sure how to stop it. Which would be hilarious if it wasn't the story of my life.

[image]

That seems to suggest 25 would be my best option? Should I try moving the ConBee II to my windows laptop to run their software and make that change?

And I truly appreciate your time and patience with me.

tube0013, 2022-10-24:
To end it just hit control-c

25 looks better then the rest. Worth a shot.

I'm not super familiar with deconz but I think you are on the right track about using their software to change the channel. You'll need to rejoin all the devices but you don't have many so hopefully not a big deal.

chicknlil, 2022-10-25:
Okay. I will be doing that shortly.

What's the proper order of operation in terms of rejoining devices back once I get there?  The living room router first?  The basement router second? Try to add the basement vibration sensor to the basement router and the kitchen vibration sensor to the living room router?

tube0013, 2022-10-25:
Add the routers first as they should mesh through each other and the coordinator. Then add the sensors.

The more the merrier 

[image]

chicknlil, 2022-10-25:
Is this more like it should be?

[image]

Now if I can just figure out how to make those vibration sensors work right. All I've done with them so far is run them out of batteries!

tube0013, 2022-10-25:
yes. that looks good. 

the vibrations sensor will show a clear or detected sensor.

you can change the sensitivity (documents in this forum thread <https://community.home-assistant.io/t/zha-xiaomi-aqara-vibration-sensor-settings/152917>

there are events for tilt and drop, maybe check for blue prints there.

oh some good stuff there:

https://community.home-assistant.io/t/zha-aqara-vibration-sensor-set-sensitivity/414810

https://community.home-assistant.io/t/blue-print-to-work-with-aqara-vibration-sensor-for-automation-to-trigger-notification-or-other-action-when-dishwasher-or-washing-machine-is-done/282553

chicknlil, 2022-10-25:
I've got that last one (the blueprint). It mentions that it needs to have a device_class: vibration, and I don't **think** I've ever seen that. But I may not know what I'm looking for? (as I've proved in this thread!). The dishwasher has run twice since I've installed the sensor, the washing machine as well (today though was just it's monthly cleaning tablet). The automations never ran. I've already replaced the battery on the dishwasher (!!!) and it looks like I now need to on the washing machine.

chicknlil, 2022-10-25:
Do you have a way I can reach you outside of github? I want to show you something, but not on here.

tube0013, 2022-10-25:
store @ tubeszb.com works

closed 2022-10-25 by chicknlil