tube0013/tube_gateways #103: Loosing connection

Sineos, 2022-11-08:
zigbee2mqtt is loosing connection to the adapter every once in a while (~ every 3-4h) and reconnects to it after about 10s:

```
  zigbee-herdsman:adapter:zStack:znp:log Socket error +2h
  zigbee-herdsman:adapter:zStack:znp:log Port closed +3ms
  zigbee-herdsman:controller:log Adapter disconnected' +22s
Zigbee2MQTT:error 2022-11-07 21:29:51: Adapter disconnected, stopping
Zigbee2MQTT:debug 2022-11-07 21:29:51: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-11-07 21:29:51: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
Zigbee2MQTT:info  2022-11-07 21:29:51: Disconnecting from MQTT server
```

Any chance that this might improve with newer esphome? I'm using https://github.com/tube0013/tube_gateways/tree/main/models/retired/tubeszb-cc2652-poe

If so, would this be the correct file: https://github.com/tube0013/tube_gateways/blob/main/models/retired/tubeszb-cc2652-poe/firmware/esphome/tube-zb-gw-cc2652p2-poe_olimex_round_latest.bin

Just drop it to the OTA filed?

tube0013, 2022-11-08:
Try the bin from the beta folder, which is the working out pretty well.

Yes you can update via the OTA section on the web front page. 

If still persists I'd try a full erase/flash by doing it over the usb port that you can access if you remove the lid and partially remove the board from the case.  Do not leave PoE connected while using USB.

Sineos, 2022-11-08:
Thanks a lot. When doing the "full" upgrade via USB
* Use https://github.com/esphome/esphome-flasher for flashing?
* Anything to backup beforehand?
* Any risk of having to pair all devices afterwards?

tube0013, 2022-11-08:
The esp32 is just a "tunnel" to the zigbee module. No backup or risk to the zigbee side.

Sineos, 2022-11-09:
Did a full flash and installed the beta. So far no more disconnects compared to regular ones every 3h to 4h. Thanks a lot

closed 2022-11-09 by Sineos