tube0013/tube_gateways #104: Power issue

astappiev, 2022-11-16:
Hi, first of all thank you for the great project.

I got mine CC2652P2 ethernet USB coordinator from ebay (b-ware) in Germany, but unfortunately it comes with a partially broken USB port (I think).

The issue is that after some time of work it turns off. After reconnecting or moving the USB it works again. There are no visible problems with the USB connector and of course I tried a bunch of different cables.

Is there any other easy way to power the board?
I see 3.3v, GND holes on the bottom left, are they input or output?

Thanks again!

tube0013, 2022-11-16:
There should be 3.3v and ground pins accessible which you could use as alternative power via a usb serial adapters power and ground pins.

Depending on condition of the pads for the usb port a cellphone repair shop or similar may be able to fix it.

astappiev, 2022-11-17:
Thank you for the quick reply, I will try to use the pins for power.
Will update with the results later

astappiev, 2022-12-27:
As you suggested, I looked for USB adapters I have, but the only thing I found was a battery charger like below. Which I ugly soldered to the very first transistor's pins near the USB port on the gateway.
[image]

But later replaced the MicroUSB port, which I ordered from Aliexpress ([a box with 10 different models](https://www.aliexpress.com/item/1005003311571316.html), one was similar). Likely all work now, including data pins :)

closed 2022-12-27 by astappiev