tube0013/tube_gateways #105: coordinator losing signal if I move it to the next room

ranrinc, 2022-11-20:
I need some advice and solutions, recently I add more and more Zigbee devices and hope that each additional bulb or switch will add more range. However, the reality is each device outside my coordinator's room seems to be unreachable. For example, all my hue bulbs work great when the coordinator is inside the room. But if I move it to the next room within an hour 2 out of 3 hues will not function anymore. Some work without a problem for example my Xiaomi button and some tuya plugs. 

Do I need a repeater? since tubezb seems to be out of stock, when will it be available? if not any alternative?

tube0013, 2022-11-20:
The bulbs and other mains powered devices act as repeaters.

It's possible there is interference at play. You can use the zigpy-cli tool to run an energy scan and check you are on the best channel for your rf environment

ranrinc, 2022-11-22:
ZHA actually comes with visualizations which I think might show the problem in connections.

In this picture you can see a red line, however, the device is actually very close to the coordinator and the rest like hue, etc did not show a red line.
[image]

This show a yellow line in which the connection is quite far. between the device.
[image]

tube0013, 2022-11-22:
Unless the devices all share the same radio type - ezsp/ti/Nordic/stm/ncp the lqi numbers don't really say much as each radio calculates and reports then differently.

If having connection issues 99/100 times interference is at work.

ranrinc, 2022-11-22:
My current problem is the house is big and I have 1 router + 5 AP with all bandwidth being deployed to make the wifi signal great. I believe this might be the culprit. Might need to rotate my WIFI channel with the Zigbee intact and see what might be a good solution for it. BTW thanks for the insight. 

Oh, when will you get more of that router available? want to purchase it

tube0013, 2022-11-22:
Still try and run the energy scan it will give insight to how your Wi-Fi is interacting with the zigbee channels.

https://github.com/zigpy/zigpy-cli

The older zigpy-znp tools have good docs for any host system to run the python tools:

https://github.com/zigpy/zigpy-znp/tools.md

[image]

tube0013, 2022-11-22:
Routers are available now.

ranrinc, 2022-12-19:
Hi @tube0013 

Since the repeater has not yet arrived and since I get a little bit more understanding of Zigbee network. Maybe you can help me out a little. My current 6 AP wifi channel is now being set to 1 and 11 which basically cover my entire house. This leaves channel 6 open for Zigbee. 

I understand that zha can be set for a fixed channel what should I set up, so it won't get lost somewhere?

```
zha:
  zigpy_config:
    network:
      channel: 15             # What channel the radio should try to use.
      channels: [15, 20, 25]  # Channel mask
```

the current setup is on 15 which from time to time will surely get tangled with my wifi channel 1
any idea on how I should set up the fixed setting so it won't get interferences?

tube0013, 2022-12-19:
Please run the energy-scan Wi-Fi is not the only source possible of interference.

ranrinc, 2022-12-26:
Hi @tube0013 

First of all Merry Christmas to you. and quick question, I'm planning to redo my Zigbee network and able to change the channel to 18 instead of 15. However, it seems my network still ignores the ZB router that I just installed. 

Also when I reset my ZHA it was set up using Hardware or Software 
which I'm not sure about. Maybe you can share some light?

[image]

tube0013, 2022-12-26:
Thanks.

How did you change the channel?

Also did you ever run a scan? 18 is intermixed even more with certain wifi channels.

ranrinc, 2022-12-27:
> Thanks.
> 
> How did you change the channel?
> 
> Also did you ever run a scan? 18 is intermixed even more with certain wifi channels.

Channel can be changed directly via HA configurations. Furthermore, channel 18 is the right choice since it's right in the middle channel which is available in my house between 1 and 11 being used for wifi. Channels on Zigbee have different numbers, so basically I set it up to fill in channel 6 on wifi for Zigbee.

Now everything works fine except 1 of my HUE bulb <- this one still become unavailable still after 24 hours even though the repeater is under 1 room. The rest seems to be working flawlessly now.

tube0013, 2022-12-27:
Cool.

The channel define in configuration.yaml for zha is only used during initial network formation if changed later it won't make any difference, that's why I asked.