tube0013/tube_gateways #109: cordinator stopped working on HA, cannot flash CC2652

tmottabr, 2022-12-06:
i am not sure how many days ago the controlled stopped working on home assistant, noticed because the lights color according to the automations.

the zigbee integration was failing on startup. I tried removing the integration and re-configuring it but now home assistant cannot connect to the controller.

I tried using upgrading to the latest yaml from here with latest esphone (2022.11.5):

https://github.com/tube0013/tube_gateways/blob/main/models/retired/tubeszb-cc2652-poe/firmware/esphome/beta/tubeszb-cc2652-poe.yaml

it is still failing. i decided to try re-flash the CC2652p2 firmware.

I tried both the latest release and devel but both fail with the same error:

```
Opening port socket://172.16.10.200:6638, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Firmware file: Intel Hex
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```
i had flashed it in the past without problem.

When i try adding the integration in home assistant the controller logs show this:

```
16:06:20 | [D] | [streamserver:074] | New client connected from 172.16.10.22
16:06:20 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
16:06:25 | [D] | [streamserver:102] | Client 172.16.10.22 disconnected
16:06:25 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
```

not sure where i can get the HA logs showing it trying to add the zigbee integration, core logs do not show anything.

tube0013, 2022-12-06:
try A flash of the esp32 /esphome fw you linked above using the usb port on the module, accessible inside the case, you may need to lift the board out to use. be sure not to have PoE plugged at the same time.

you can use the esphome flasher tool - this will do a clean flash- a full erase first before the actual flash.

tmottabr, 2022-12-06:
done.. still same behavior after.. home assistant cannot connect and cannot flash cc2652..

tube0013, 2022-12-06:
can you try running a zigpy-cli energy-scan to see if the connection issue presents with it:

https://github.com/zigpy/zigpy-cli 

if it starts it will run until stopped with control-c

you can also try a flash, via network by unplugging and holding the physical BSL button while plugging in the PoE. continue holding for about 15 seconds so the esp32 boots as it is set to reset the zigbee module on boot as well. 

but if nothing is reliably connecting the serial it could be hardware issue. but let's see what happens with above.

tmottabr, 2022-12-07:
unfortunately not even zigpy is connecting.. i am getting a timeout, asyncio.TimeoutError, when i try doing anything with it.. 

booting with BSL also did not changed anything, still same behavior for everything..

thanks a lot for the assistance, but if this is a hardware problem ill be looking into getting a new one device soon..

tube0013, 2022-12-07:
Seems that way. Please reach out to me at store@tubeszb.com and will see what I can do.