tube0013/tube_gateways #110: Unable to configure Zigbee to PoE Coordinator in Home Assistant

joeypiccola, 2022-12-07:
**Home Assistant**
Version: `2022.10.4` (the one that runs as a container)
Network: `10.0.3.0/24 (server VLAN)`

**Tubeszb**
Device: CC2652P2 Based Zigbee to PoE Coordinator 2022
Network: `192.168.9.0/24 (IoT VLAN)`

There's full routing between the two networks mentioned above, there are no restrictions. 

Home Assistant (HA) discovers the Tubeszb device. However, when I attempt to configure it I get "Error" in the HA UI. 

Discovery:
[image]

Dialog when I click configure:
[image]

Confirm serial port settings:
[image]

Network formation: I have selected both "Keep radio network settings" and "Erase network settings and form a new network".
[image]

Dialog once a network formation is selected:
[image]

The Error:
[image]

tube0013, 2022-12-07:
Can you please try to see if the device connects when on the same vlan as the server, then we can go from there to try and troubleshot the vlans.

joeypiccola, 2022-12-07:
Unfortunately, still the same issue.

joeypiccola, 2022-12-07:
Logging enabled.

```yaml
logger:
  default: info
  logs:
    zigpy: debug
    zigpy_znp: **debug**
```

```plaintext
2022-12-07 15:57:32.105 DEBUG (MainThread) [zigpy.appdb] Current database version is v10
2022-12-07 15:57:32.106 DEBUG (MainThread) [zigpy.appdb] Loading application state
2022-12-07 15:57:32.114 DEBUG (MainThread) [zigpy_znp.uart] Connecting to socket://10.0.3.51:6638 at 115200 baud
2022-12-07 15:57:32.116 DEBUG (MainThread) [zigpy_znp.uart] Opened socket://10.0.3.51:6638 serial port
2022-12-07 15:57:32.116 DEBUG (MainThread) [zigpy_znp.uart] Connected to socket://10.0.3.51:6638 at 115200 baud
2022-12-07 15:57:32.116 DEBUG (MainThread) [zigpy_znp.api] Toggling RTS/DTR pins to skip bootloader or reset chip
2022-12-07 15:57:32.571 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2022-12-07 15:57:32.632 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-12-07 15:57:32.633 DEBUG (MainThread) [zigpy_znp.api] Radio is alive: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2022-12-07 15:57:32.633 DEBUG (MainThread) [zigpy_znp.nvram] Detecting struct alignment
2022-12-07 15:57:32.633 DEBUG (MainThread) [zigpy_znp.api] Sending request: UTIL.AssocFindDevice.Req(Index=0)
2022-12-07 15:57:32.704 DEBUG (MainThread) [zigpy_znp.api] Received command: UTIL.AssocFindDevice.Rsp(Device=b'\xFE\xFF\x00\x20\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF')
2022-12-07 15:57:32.706 DEBUG (MainThread) [zigpy_znp.nvram] Detected struct alignment: True
2022-12-07 15:57:32.706 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=0)
2022-12-07 15:57:32.777 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.NVLength.Rsp(Length=20)
2022-12-07 15:57:32.778 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVRead.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.TCLK_TABLE: 4>, SubId=0, Offset=0, Length=20)
2022-12-07 15:57:32.850 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.NVRead.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00')
2022-12-07 15:57:32.851 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM[NvSysIds.ZSTACK][ExNvIds.TCLK_TABLE][0x0000] = b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xFF\x00\x00\x00'
2022-12-07 15:57:32.851 DEBUG (MainThread) [zigpy_znp.api] Detected Z-Stack 3.3
2022-12-07 15:57:32.851 DEBUG (MainThread) [zigpy_znp.api] Connected to socket://10.0.3.51:6638
2022-12-07 15:57:32.852 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.NIB: 33>)
2022-12-07 15:57:32.923 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=116)
2022-12-07 15:57:32.923 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.NIB: 33>, Offset=0)
2022-12-07 15:57:32.997 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVReadExt.Rsp(Status=<Status.SUCCESS: 0>, Value=b'\x2B\x05\x02\x33\x0F\x33\x00\x1E\x00\x00\x00\x01\x05\x01\x8F\x00\x07\x00\x02\x0D\x1E\x00\x00\x00\x0F\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xBD\x2D\x08\x00\x00\x80\x00\x00\x0F\x0F\x04\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\xBF\x9A\xFC\x3A\x91\xF1\x9A\x24\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0F\x05\x00\x01\x64\x0A\x01\x00\x00\x00\x00\x00\x00\x00')
2022-12-07 15:57:32.999 DEBUG (MainThread) [zigpy_znp.nvram] Read NVRAM["LEGACY"][0x0021] = NIB(SequenceNum=43, PassiveAckTimeout=5, MaxBroadcastRetries=2, MaxChildren=51, MaxDepth=15, MaxRouters=51, dummyNeighborTable=0, BroadcastDeliveryTime=30, ReportConstantCost=0, RouteDiscRetries=0, dummyRoutingTable=0, SecureAllFrames=1, SecurityLevel=5, SymLink=1, CapabilityFlags=143, TransactionPersistenceTime=7, nwkProtocolVersion=2, RouteDiscoveryTime=13, RouteExpiryTime=30, nwkDevAddress=0x0000, nwkLogicalChannel=15, nwkCoordAddress=0x0000, nwkCoordExtAddress=00:00:00:00:00:00:00:00, nwkPanId=11709, nwkState=<NwkState.NWK_ROUTER: 8>, channelList=<Channels.CHANNEL_15: 32768>, beaconOrder=15, superFrameOrder=15, scanDuration=4, battLifeExt=0, allocatedRouterAddresses=1, allocatedEndDeviceAddresses=1, nodeDepth=0, extendedPANID=24:9a:f1:91:3a:fc:9a:bf, nwkKeyLoaded=<Bool.true: 1>, spare1=NwkKeyDesc(KeySeqNum=0, Key=00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00), spare2=NwkKeyDesc(KeySeqNum=0, Key=00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00), spare3=0, spare4=0, nwkLinkStatusPeriod=15, nwkRouterAgeLimit=5, nwkUseMultiCast=<Bool.false: 0>, nwkIsConcentrator=<Bool.true: 1>, nwkConcentratorDiscoveryTime=100, nwkConcentratorRadius=10, nwkAllFresh=1, nwkManagerAddr=0x0000, nwkTotalTransmissions=0, nwkUpdateId=0)
2022-12-07 15:57:32.999 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVLength.Req(Id=<OsalNvIds.BDBNODEISONANETWORK: 85>)
2022-12-07 15:57:33.069 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.OSALNVLength.Rsp(ItemLen=1)
2022-12-07 15:57:33.070 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.OSALNVReadExt.Req(Id=<OsalNvIds.BDBNODEISONANETWORK: 85>, Offset=0)
[... cut ...]
2022-12-07 15:58:06.776 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.NVWrite.Rsp(Status=<Status.SUCCESS: 0>)
2022-12-07 15:58:06.778 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.NVLength.Req(SysId=<NvSysIds.ZSTACK: 1>, ItemId=<ExNvIds.ADDRMGR: 1>, SubId=79)
2022-12-07 15:58:06.800 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.ResetReq.Req(Type=<ResetType.Soft: 1>)
2022-12-07 15:58:06.800 DEBUG (MainThread) [zigpy_znp.api] Request has no response, not waiting for one.
2022-12-07 15:58:06.801 DEBUG (MainThread) [zigpy_znp.uart] Closing serial port
```

tube0013, 2022-12-07:
so all is going well, then the serial connections drops.

are you comfortable running some python command line tools?

if so can you try https://github.com/zigpy/zigpy-cli

I'd run a reset, followed by a form, and then an energy scan and see if those commands work.

```
zigpy radio znp socket://10.9.3.51:6638 reset
zigpy radio znp socket://10.9.3.51:6638 form
zigpy radio znp socket://10.9.3.51:6638 energy-scan
```

tube0013, 2022-12-07:
oh - for the first 2, use the -vv switch for some feedback (it will look similar the ha logs)

```
zigpy -vv radio znp socket://10.9.3.51:6638 reset
zigpy -vv radio znp socket://10.9.3.51:6638 form
zigpy radio znp socket://10.9.3.51:6638 energy-scan
```

tube0013, 2022-12-07:
^ ignore this then.

joeypiccola, 2022-12-07:
Is it of concern that both of these integrations show up in HA?

[image]

tube0013, 2022-12-07:
okay. all this is good. as it shows device is working as expected.

you can kill the scan.

I saw in the log it was restoring a backup - is this the case?

joeypiccola, 2022-12-07:
> I saw in the log it was restoring a backup - is this the case?

Not to my knowledge, when setting it up the only setting I have selected have been "Keep radio network settings" and "Erase network settings and form a new network".

tube0013, 2022-12-07:
okay, well there is a networked now formed on the coordinator from the testing (the form command), so you should be able to add zha now and pick keep radio settings.

give that a try and hopefully the setup goes through

joeypiccola, 2022-12-07:
Well, before I read your last comment I wanted to try the following. 

My HA instance is accessed via a reverse proxy (specifically HAProxy). To remove any issues with that I accessed my HA instance via the direct URL (e.g. `http://10.0.3.111:8123/`). I went through the configure routine (selecting "Erase network settings and form a new network") and it worked. 

I will need to move the coordinator back to the IoT VLAN. Later this evening I will remove the integration, move the coordinator back to the IoT VLAN, and attempt setup again via a direct URL to my HA instance.

I'm not sure why doing the configuration of the coordinator via my proxy address, `https://ha.mydomain.com/`, causes issues.

tube0013, 2022-12-07:
hmm. yeah.

so when you move it, all you need to do is do a migrate > re-configure current radio and then change the address.

joeypiccola, 2022-12-08:
Moved to the new VLAN\IP address. Still works. Thank you for all the help.

tube0013, 2022-12-08:
Great. I wonder if it got into a funky state on the initial attempts and the testing- reset etc helped clear things out.

closed 2022-12-08 by joeypiccola

Sleashe, 2023-02-07:
I confirm I had the same issue. Accessing HA from a reverse proxy ends to an issue when adding TubeZB via auto or manual integration. Accessing HA from IP resolved everything, no matter the VLAN.

I didn't take to read ZHA code, but I think TubeZB logic and HA are communicating their backlinks to each other when starting association process.

In my case, pfSense was not allowing IOT devices (Where TubeZB is) to access my reverse proxy instance. That's why integration was failing.

Aephir, 2023-05-04:
Not to poke a dead horse, but I came upon this issue in my troubleshooting, so in case others have the same issue as me:

Nothing worked, regardless of VLAN, local IP vs. NGINX-served FQDN, and nothing found with zigpy-cli seemed out of the ordinary.

The only thing that made a difference was killing my deCONZ docker container and `disabling` the deCONZ integration in HA. Then after a HA restart, I could integrate the CC2652P2 PoE Coordinator with ZHA with absolutely no issues.