tube0013/tube_gateways #111: ESP 32 Flash on tubeszb-cc2652-eth_usb + WT32-ETH01 flashing problem...

mcarbonneaux, 2022-12-07:
i've buy 2 tubesb cc2652 eth usb coordinator.
i've buy separatly 3 WT32-ETH01 (v1.4)

i followed the instructions : https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#esp32-flashing

i've installed the CH340 driver
the com port are found by windows (in device manager).
i've downloaded esphome-flasher (v1.4.0) 32bit and 64bit windows version.
i've downloaded the esp32 firmware : https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-cc2652-eth_usb/firmware/esphome/tube-zb-gw-cc2652p2-v2_2021_09_17.bin
but when i try to flash... i take an error:
```
Using 'COM5' as serial port.
Connecting......................................
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
```
when i click on logs:
```
Using 'COM5' as serial port.
Unexpected error: could not open port 'COM5': PermissionError(13, 'Accès refusé.', None, 5)
```

i've tried to change my cable by another one.... the same...

i'm on Windows 10 22H2 (i'm on ryzen 5 2600 + gigabyte mb B450)...
the version of the driver CH340 : 3.5.2019.1

tube0013, 2022-12-07:
are you holding the EBOOT button down when plugging in the device to flash? (you can also hold it and tap the ERST button to reset the esp32 into bootloader mode).

tube0013, 2022-12-07:
did the boards work with zigbee over usb without issue?

mcarbonneaux, 2022-12-07:
erst tryed and not permit to flash...

" zigbee over usb " how to try that ?

tube0013, 2022-12-07:
hook up to home assistant server with usb cable?

do the wt32-eth01 modules light up when plugged into USB? you should see a red LED.

mcarbonneaux, 2022-12-07:
ok i've done that : press and hold RESET, press BOOT, release Reset and then release BOOT,
and flash are gooing !

mcarbonneaux, 2022-12-07:
```
Auto-detected serial port: COM5
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ5 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: xx:xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.1s
Compressed 15872 bytes to 10319...
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.3 seconds (effective 501.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 3619.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 16381.3 kbit/s)...
Hash of data verified.
Compressed 924768 bytes to 524202...
Wrote 924768 bytes (524202 compressed) at 0x00010000 in 13.5 seconds (effective 547.0 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!
```

mcarbonneaux, 2022-12-07:
is needed to flash also the ti part (CC2652) or is pre flashed?

closed 2022-12-07 by mcarbonneaux

tube0013, 2022-12-07:
The cc2652 is pre-flashed.

mcarbonneaux, 2022-12-07:
> erst tryed and not permit to flash...
> 
> " zigbee over usb " how to try that ?

ok is to use the coordinator usb zigbee dongle for z2m or zha ?!

tube0013, 2022-12-07:
you can go straight to ethernet now if you have the wt32-eth01 flashed and installed. 

guides are on the site and z2m has docs for setting up a network connected coordinator (skip to step 3 on that page).

mcarbonneaux, 2022-12-08:
i while test that tomorow ! :)