tube0013/tube_gateways #113: it's possible to flash TubesZB CC2652 Ethernet/USB Coordiniator with router firmware ?

mcarbonneaux, 2022-12-18:
at step:
- https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#cc2652-firmware-flashing

i can use router version in place ? like :
- https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0/bin

and with what version ? :
- [CC1352P2_CC2652P_launchpad_router_20220125.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20220125.zip)
- [CC1352P2_CC2652P_other_router_20220125.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_router_20220125.zip)

tube0013, 2022-12-18:
Yes. Note the Ethernet will okay no role in extending the zigbee networks

You use the launchpad variant or the build I use on the routers:

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-router/firmware

mcarbonneaux, 2022-12-18:
no problem with the ethernet, i understand.

i can use your last date version : [tubeszb_router_20221002.hex.zip](https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-router/firmware/tubeszb_router_20221002.hex.zip) ?

tube0013, 2022-12-18:
> no problem with the ethernet, i understand

Only stated as it's often misunderstood.

Yes on the version. Been shipping routers with that version for a while.

mcarbonneaux, 2022-12-18:
if i whant tu reuse in coordinator mode i juste need to reflash with coordinator firmware ?

tube0013, 2022-12-18:
Yes, going back is the same process.

mcarbonneaux, 2022-12-18:
ok cool! for the moment my need is more router than additional coordinator.

mcarbonneaux, 2022-12-18:
it's possible to build 20db version of the router or is counter productive ?

closed 2022-12-18 by mcarbonneaux

tube0013, 2022-12-18:
You may want to look at the dev version of the router in the z-stack repo, it has adjustable tx strength.

Will it be able to hear everything that hears it?

mcarbonneaux, 2022-12-18:
i've adjusted my usb coordinator based on CC2652P (sonof) to 15db (it's configurable in zigbee2mqtt) but it's emission power not receive power, and if router as the power emission positionned to 15db too i'm abel to make bridge between too long distance part of my home.

mcarbonneaux, 2023-01-09:
it's work fine as router !

what is the sequence to force reapairing ?

tube0013, 2023-01-09:
momentarily press the BSL button either with esphome through the web dash board or with the physical button on the coordinator.