tube0013/tube_gateways #114: Can not pair aqara devices to efr32 POE

raphael303, 2022-12-22:
I tried channels 11, 15, 25 while I removed my 2.4 Wlan to the other side of the spectrum. I can pair IKEA devices and the Aqara can connect to the IKEA devices if they are close enough and routers, but the aqara devices do not connect directly to the EFR32. 

In fact they will pick a router twice as far away, but never connect to the EFR32. Not even when every other device is off.

tube0013, 2022-12-22:
If you go to the coordinator device and say pair only with this device does it work?

raphael303, 2022-12-22:
I don't know how I would do that. I'm using homeassistant and zigbee2mqtt. 

Everything else works fine since I changed the channel to 25 and have all the wlan bellow channel 10. Today I got a bunch of aqara devices in the mail and those don't work with direct connections as described above.

tube0013, 2022-12-22:
Okay I only know how to do it with ZHA. EFR32 is experimental with Zigbee2mqtt, so I can't support that. If they work with routers join them directly to them.

tube0013, 2022-12-22:
run an energy scan. all that sounds like interference. - https://github.com/zigpy/zigpy-cli

with zha logging at least you can see the ezsp counters and see if packets are refusing to send because of interference. si

raphael303, 2022-12-22:
Thank you for the tip. But I don't even know where to run this energy scan. Do I need to setup an extra raspberry with raspian?

tube0013, 2022-12-22:
You can use any device that you can install python on to run the tool. The precursor to this zigpy-cli has a great guide to setting up python on any platform - https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

You can stop before the install zigpy-znp step, and instead pick up with the pip command to install zigpy-cli.
 to address the network connected coordinator with zigpy-cli you use socket://ip_address:6638 - sample energy scan command below:

zigpy radio znp socket://[192.168.1.123:6638](http://192.168.1.123:6638/) energy-scan

raphael303, 2022-12-22:
Thank you. I tried. I really did. I could install everything  in you links, no errors, all went fine.
Then when I do the energy scan it fails.

''' zigpy.application ERROR Couldn't start application
Traceback (most recent call last):
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 664, in _skip_bootloader
    result = await responses.get()
             ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.0/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy/application.py", line 131, in startup
    await self.connect()
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 106, in connect
    await znp.connect()
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 706, in connect
    self.capabilities = (await self._skip_bootloader()).Capabilities
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.0/Frameworks/Python.framework/Versions/3.11/lib/python3.11/contextlib.py", line 222, in __aexit__
    await self.gen.athrow(typ, value, traceback)
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 866, in capture_responses
    yield queue
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 663, in _skip_bootloader
    async with async_timeout.timeout(CONNECT_PROBE_TIMEOUT):
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
Traceback (most recent call last):
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 664, in _skip_bootloader
    result = await responses.get()
             ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.0/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/raphaelhuber/venv/bin/zigpy", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/click/decorators.py", line 38, in new_func
    return f(get_current_context().obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_cli/cli.py", line 19, in inner
    return loop.run_until_complete(cmd(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.0/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/base_events.py", line 650, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_cli/radio.py", line 178, in energy_scan
    await app.startup()
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy/application.py", line 131, in startup
    await self.connect()
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 106, in connect
    await znp.connect()
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 706, in connect
    self.capabilities = (await self._skip_bootloader()).Capabilities
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/Cellar/python@3.11/3.11.0/Frameworks/Python.framework/Versions/3.11/lib/python3.11/contextlib.py", line 222, in __aexit__
    await self.gen.athrow(typ, value, traceback)
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 866, in capture_responses
    yield queue
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/zigpy_znp/api.py", line 663, in _skip_bootloader
    async with async_timeout.timeout(CONNECT_PROBE_TIMEOUT):
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/Users/raphaelhuber/venv/lib/python3.11/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
(venv) raphaelhuber@Raphaels-Compiutr ~ % 
'''

I have disabled MQTT2Zigbee on HA, and I have restarted the efr32.

raphael303, 2022-12-22:
zigpy radio znp socket://192.168.1.151:6638 energy-scan
the IP is correct. ping works.

tube0013, 2022-12-22:
Go to the coordinator ip in a browser, be sure nothing is connected to serial.

And trigger a efr32 reset

Then try again

raphael303, 2022-12-22:
thank you for your help.
Did that. nothing is connected, reset, didn't work. Same errors. Tried the command with sudo too.

tube0013, 2022-12-22:
does z2m still start?

raphael303, 2022-12-22:
Yes it starts.

I repeated everything on homeassistant, with the SSH tool there.
Same error. 
In the webinterface of the tubes I see this:

```
Time | level | Tag | Message
-- | -- | -- | --
20:54:36 | [D] | [streamserver:074] | New client connected from 192.168.1.103
20:54:36 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
20:54:46 | [D] | [streamserver:102] | Client 192.168.1.103 disconnected
20:54:46 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state O

```

And in the ssh of ha I see pretty much the same as before.

tube0013, 2022-12-22:
hmm. something seems off then, if z2m can connect the tool should. can you try with python 3.10? 

or post that trace in an issue on the zigpy-cli repo.

raphael303, 2022-12-22:
Thank you very much for your help, I really appreciate it. But at this point I give up.

I have invested days and days in this zigbee stuff. At this point I must draw a line. I wait a couple of months and hope that updates will solve this. Or that with HA yellow arriving I will use that zigbee radio and hope it works better, even though its the same chip. little hope though. 

ZHA really doesn't cut it for me. zb2mqtt is not supported. So end of line here. No more precious time can go into this.

I have done a wifi scan with wifiexplorer and everything is as I set it up. There is no wifi here but mine, the next houses are 100m away. 

All my 4 APs are bellow or at channel 9, the ones that are at channel 5 and 9 are farthest away from anything zigbee here. Zigbee is at channel 25. Even if there is an interference problem, I would not know how to do anything different.

Again thank you for your support. I have learned some stuff which I will hope will help me solve other problems and maybe this exchange may help others or point out a bug in zigpy-cli.

tube0013, 2022-12-22:
there are more things than wifi that can interfere with zigbee that's why I always suggest an energy scan. you can also try and use bellows to do the scan but it won't be a nice of a display.

this is a way to do it via docker > https://pastebin.com/bk1hN0Et

but you can just do pip isntall bellows on your mac (better yet create a py venv as the zigpy-znp tools.md shows) then run 

 bellows -b 115200 -d socket://192.168.1.151:6638 scan -e

raphael303, 2022-12-22:
[11, -65]
[12, -46]
[13, -46]
[14, -66]
[15, -56]
[16, -52]
[17, -76]
[18, -72]
[19, -72]
[20, -63]
[21, -72]
[22, -78]
[23, -90]
[24, -90]
[25, -87]
[26, -61]

raphael303, 2022-12-22:
I guess -87 is something like signal to noise? only channel 23/24 would be even better than 25.

tube0013, 2022-12-23:
Lower is better so 25 likely the best.

raphael303, 2022-12-30:
In the end I gave up on this, bought a sonoff CC2652P dongle because I could get it within a day. I re-paired everything and without any hassle it all works now.

So, bottom line: EFR32 and ZB2MQTT  or/and aqara and some ikea devices doesn't work. I know it is "experimental", but I'd rather say it plain does not work.

Still think the tubes device is good though. It was my mistake to want the "future proofer" efr32... I don't recommend it to anyone who is not using zha.

tube0013, 2022-12-30:
Closing this. Appreciate the feedback.

closed 2022-12-30 by tube0013