tube0013/tube_gateways #120: Routers offline after initial join

stevewull, 2023-01-30:
Hello,
I just purchased four of the router/repeaters. When plugged in, they successfully joined my Zigbee network. But within 15 minutes, each of them went offline.  Is there a setting I need to make?

Coordinator: SLZB-06
Home Assistant
Zigbee2MQTT add-on

[image]

tube0013, 2023-01-30:
Can you try and read an attribute from the. Basic cluster to see if they respond.

stevewull, 2023-01-30:
How do I do that in Z2M?

stevewull, 2023-01-30:
Hmmm. Now two of them are back online...

tube0013, 2023-01-30:
I always forget how to do it in z2m, I'll have to set up my test environment later tonight to double check.

In other cases the devices would show offline in the z2m UI, but still be functioning.

stevewull, 2023-01-30:
Could be.  It is saying that z2m failed to ping the devices and therefore marks them as offline. I'll start to test whether my endpoints connect to the router anyway.

Failed to ping 'Garage Router' (attempt 1/1, Read 0x00124b002591007e/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))

tube0013, 2023-01-30:
Yeah, I'm not familiar with the z2m "ping" processes. thought based on above it looks to just try and read an attribute - zclVersion from the basic cluster.

it definitely should respond back to that. do you have an USB to serial converter? you could try the previous fw I have on github from 02-2022. and compare.

stevewull, 2023-01-30:
Yes. I have a USB to serial converter.

tube0013, 2023-01-30:
instructions are here:
https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-router#flashing-the-router

and this is the earlier fw:
https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-router/firmware/tubeszb_router_20220223.hex.zip (pulll down zip and extract the hex to use for flashing)

tube0013, 2023-01-30:
Pretty much.

Windows drivers are only needed if the converter you have isn't recognized. It's really dependent on the chipset used.

Zigpy-znp has a great how to for getting python up and running in multiple OSes.

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

stevewull, 2023-01-30:
meanwhile. I placed a motion sensor next to the "Garage Router" and joined it to the network.  Its on the network and according to the Network Map, it is communicating to the Garage Router and the Garage Router is communicating to the controller, even though the router still shows offline. Motion sensor is reporting successfully.

I suppose there is a cluster somewhere that tells me which device the motion sensor is connected to.

tube0013, 2023-01-30:
The routing device it's connected to actually sends its list of children when polled for that.

stevewull, 2023-01-30:
Which means that the router is on the network and working.

Just the availability ping that doesn't always get a response.

stevewull, 2023-01-31:
Thanks for the help.  For now, I can live with the offline message.  Not sure that all the time to reflash something that is functioning fine will rise to the top of the priority list.

Is there a way to force devices like bulbs to try to find the nearest router?  Or, if they follow the Zibee spec properly (unlike Agara), will they work it out on their own?

And OT: What is the difference between the CC2652P and CC2652P2 chips?

tube0013, 2023-01-31:
sounds good.
Pretty sure CC2652P2 is the official chip variant name, and CC2652P is just short hand

stevewull, 2023-01-31:
As update. All of my devices chose to link through your routers rather than the controller even when the controller was much closer. 

all children of the routers show up as offline in z2m too.  I will try the older firmware at some point. maybe over the weekend.

doddys, 2023-02-05:
I am also having the same problem with my zigbee router

tube0013, 2023-02-05:
> I am also having the same problem with my zigbee router

Per this comment are you able to try downgrading the fw?

https://github.com/tube0013/tube_gateways/issues/120#issuecomment-1409423093

doddys, 2023-02-06:
I was looking at my home assistant history of the router lqi values. It appears that it gives a response only on the hour. 

My understanding at the moment is like this:
When The default config value for reporting is the following:
min rep interval = 0
max rep interval = 3600
min  rep change = 0
attribute = zclVersion

Tubes Router will not response when the value of the attribute does not change until the max rep interval is exceeded. 
By Default Z2M will ping the router every 10 minutes by getting the zclVersion attribute value. 

As result, after the first call on the hour, you will get online status and it will be offline until the next hour. 

What I do now to fix this is to change the max rep interval to be less than Availability - Active Timeout setting 
Active Timeout Setting = 10 minutes
max rep interval = 300 (5 minutes)

So far, my router status is now online and stable. Let me know if there is any negative impact for these settings.

stevewull, 2023-02-07:
@doddys  Thanks for the tip. I set my max rep interval to 300 and, so far, my router status is also online and stable.

tube0013, 2023-02-07:
Thanks both. I'm trying to look into this more as well but appreciate the work around suggestion!

stevewull, 2023-02-11:
Looks like the 20221002 firmware is installed.  Should I upgrade the routers to the 20221226 release?  Where are those files?

tube0013, 2023-02-11:
20221226 is coordinator fw. 

The newest router fw is in the development branch so not officially released

stevewull, 2023-02-14:
You may want me to start a new issue for this....
In addition to the routers going offline (unless I adjust the rep interval), my Sengled light bulbs connected to the mesh network through the Tube routers also go offline and become uncontrollable. I have to power them off and power back on to get them back. Then they only last up to a day before dropping off again.

I have 30 such bulbs:
Z01-A19NAE26: 18
E13-N11: 6
E11-G13: 2
E11-N1G: 2
E1F-N5E: 2

If I unplug all four of my Tube routers, such that I only have a controller and 4 Sonoff S31 zb plug repeaters, then the bulbs no longer drop off the network.  (Note though: I am conducting further tests on this to confirm.)