tube0013/tube_gateways #125: How to turn on "Prep for firmware update mode"

alex-savin, 2023-02-04:
Hey,

Thank you for your amazing devices!

I updated my devices with the new firmware and today wanted to update coordinator firmware, but could find the toggle in the new WebUI
Can you please help me to find the way to turn the "Prep for firmware update mode" with the new WebUI

Thank you in advance,
Alex

tube0013, 2023-02-04:
Hi,

in the newer esphome versions, this is just a button action to trigger the script for bootloader mode. so you just click the button.

The script just turns on the BSL toggle and preforms a reset and then a delay for about 10 seconds and prints a message to the console. You can trigger the toggles manually in the web intercase to do this same action. You can also use the physical BSL button on the PCB and hold it down while giving the device power.

alex-savin, 2023-02-04:
@tube0013 Thank you!

closed 2023-02-04 by alex-savin