tube0013/tube_gateways #126: Firmware update cadence?

nareddyt, 2023-02-05:
Hi, looks like there is a new z-stack firmware released last week: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Do you know when the firmwares for tubes ZB routers will be updated? https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-router/firmware

Thanks in advance!

tube0013, 2023-02-05:
Router fw is on different track than coordinator. The z-stack router fw all works on TubesZB device, my custom version only  modify the branding and enable an led always to be on for showing power state.

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0

There is currently a newer version in the development branch still under evaluation.

nareddyt, 2023-02-05:
Ah right. Thanks!

closed 2023-02-05 by nareddyt