tube0013/tube_gateways #127: CC2652P2 PoE Coordinator Error Lights

mttwong, 2023-02-07:
Following a power outage, my PoE coordinator does boot up to connect to the network, and just shows a solid orange light.  I looked through the documentation and can't find any information about what different light codes mean.  Any idea what could have happened?

tube0013, 2023-02-07:
The red and yellow leds that light up when it has power. There is a green led that lights when an Ethernet connection is established.

Can you see if the it connects successfully if connected to non powered Ethernet and powering the board by USB. (Have to partially lift the board out of the case to access the port. Do not use powered Ethernet at the same time as usb and PoE voltages are not isolated.)

mttwong, 2023-02-08:
I was able to get it to connect to my network with the USB powering it once, then when I moved it back to my other room and tried again, it didn't work that time.  I've tried 3 different USB wall warts and 2 different POE injectors.  I thought maybe it was only getting enough power to turn on the red and yellow lights, but not enough to start connecting to the network.

tube0013, 2023-02-08:
Can you send an email to store @ tubeszb.com please include the name you used and email used for the order. And I'll try and work something out. 

Thanks

tube0013, 2023-02-08:
I responded to the email, let me know if it didn't come through, maybe check spam ;)

closed 2023-11-15 by tube0013