tube0013/tube_gateways #128: Tubes Coordinator (POE) version failing upgrade to the December firmware update.

twinkie0101, 2023-02-07:
I tried to do the update today via bsl-master and receiving the following error after update starts:

ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'

I am updating the tubes coordinator (tubeszb-cc2652-poe)) with the following file: C1352P2_CC2652P_launchpad_coordinator_20221226.hex

tube0013, 2023-02-07:
Hi,

can you you try updating the esphome fw first, you can do this over the web OTA. if you have the round from 2021 and early 2022 please use the binary from here: 
https://github.com/tube0013/tube_gateways/tree/main/models/retired/tubeszb-cc2652-poe/firmware/esphome

If it's the 2022 module in rectangular case use the binary here:
https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-poe-2022/firmware/esphome

If you have trouble downloading the bin, you can download the zip and extract the bin file once downloaded.  after the update,

Use the web frontend and use the button to trigger the zigbee module bootloader, then attempt to use cc-2538-bsl.py again to flash the fw.

The newer esphome fw has made flashing more reliable. be prepared it takes about 5 minutes and 30 seconds to do a full flash over the network.

twinkie0101, 2023-02-08:
Did as you said and got the new esphome firmware loaded and zigbee update went smooth and can confirm working in home assistant as expected so far.  Will close for now and report back if any issues experienced.  Thanks for the quick reply!

twinkie0101, 2023-02-08:
Have had this happen a couple of times, but the coordinator just stops working (becomes inaccessible via ip address).  Not sure if these logs from Z2M are helpful but including:

```
Zigbee2MQTT:info  2023-02-08 15:49:32: Logging to console and directory: '/config/zigbee2mqtt/log/2023-02-08.15-49-32' filename: log.txt
Zigbee2MQTT:info  2023-02-08 15:49:32: Starting Zigbee2MQTT version 1.30.1 (commit #unknown)
Zigbee2MQTT:info  2023-02-08 15:49:32: Starting zigbee-herdsman (0.14.89)
Zigbee2MQTT:error 2023-02-08 15:49:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-08 15:49:51: Failed to start zigbee
Zigbee2MQTT:error 2023-02-08 15:49:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-08 15:49:51: Exiting...
Zigbee2MQTT:error 2023-02-08 15:49:51: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:198:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

tube0013, 2023-02-08:
Is the coordinator accessible over the web when this happens?

What do you have to do to get it to restart (are you able to?)

twinkie0101, 2023-02-08:
It's not, I have power cycle the port to get the device to be reachable again.

onedr0p, 2023-02-27:
> Have had this happen a couple of times, but the coordinator just stops working (becomes inaccessible via ip address). Not sure if these logs from Z2M are helpful but including:
>
> ``` [...]

I am having the same exact issue. It starting happening last week for me.

onedr0p, 2023-02-28:
Well I did update both esphome firmware and the controller firmware and now all my devices are not working. I have tried restarting the PoE controller, and zigbee2mqtt :/

```
Zigbee2MQTT:error 2023-02-27 19:56:08: Publish 'set' 'state' to 'landing__ceiling_light' failed: 'Error: Command 0x5c0272fffecdff50/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24571 - 1 - 11 - 6 - 11 after 10000ms)'
Zigbee2MQTT:debug 2023-02-27 19:56:08: Error: Command 0x5c0272fffecdff50/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24571 - 1 - 11 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:564:17)
    at processTimers (node:internal/timers:507:7)
```

twinkie0101, 2023-02-28:
Yeah, my updates unfortunately made the coordinator unusable as the device would connect but after some time (can't remember exactly, but less than 30 minutes) the coordinator dropped from my network. Reboots didn't help as well. I've migrated to a sky connect device for now.  Hoping we can figure out a fix but right now my coordinators are not stable enough to keep the family from griping about stuff not working.

onedr0p, 2023-02-28:
I think I'll have to migrate to another device as well, I know someone else who has posted here who also has this model and it suffered the same fate. He also migrated to another device. :(

twinkie0101, 2023-02-28:
My device was updated to the new esphome firmware and new zigbee firmware. Prior to making the update the coordinator was rock solid; since unstable as mentioned previously.

tube0013, 2023-03-11:
Have you tried reflashing esphome over USB?

if so and still unstable please reach out to me at store @tubeszb.com and we can discuss options.

NathanSweet, 2024-10-20:
The [comment above](https://github.com/tube0013/tube_gateways/issues/128#issuecomment-1421584114) fixed the error OP report for me. I didn't remember what version I had (2022) and using 2023 ESP firmware prevented flashing the ZB module firmware.