tube0013/tube_gateways #132: Sengeld Bulbs with cc2652 routers

tube0013, 2023-02-14:
You may want me to start a new issue for this....
In addition to the routers going offline (unless I adjust the rep interval), my Sengled light bulbs connected to the mesh network through the Tube routers also go offline and become uncontrollable. I have to power them off and power back on to get them back. Then they only last up to a day before dropping off again.

I have 30 such bulbs:
Z01-A19NAE26: 18
E13-N11: 6
E11-G13: 2
E11-N1G: 2
E1F-N5E: 2

If I unplug all four of my Tube routers, such that I only have a controller and 4 Sonoff S31 zb plug repeaters, then the bulbs no longer drop off the network.  (Note though: I am conducting further tests on this to confirm.)

_Originally posted by @stevewull in https://github.com/tube0013/tube_gateways/issues/120#issuecomment-1429041307_

stevewull, 2023-02-14:
Also as an FYI for anyone reading this, Sengled publishes an FAQ about use conditions for its smart bulbs at: https://support.sengled.com/hc/en-us/articles/1260803627550-Troubleshooting-Sengled-Smart-LED-ZigBee-Bulb

tube0013, 2023-02-14:
@stevewull I set up another issue for this.

Are you able to try the koenkk's main router fw, (I am just seeing he recently moved the 20221102 developer version to master).  But I'd be interested if you experience the same issues. https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0

I've had another customer a few months back where they experienced similar with Sengled Bulbs. I myself am not able to use Sengled Bulbs as they do not stay on my mesh. We spent a lot of time troubleshooting this with network sniffs and talking with the main zha devs. Our final determination was that there is a random mix of devices on a network where this issue presents. The sniffs showed the Sengled devices creating a lot of NWK conflicts, and then giving up trying to re-join as they should in that situation so we were also not convinced there wasn't a flaw in the Sengled FW.

During the testing I set up a test network with 2 or 3 Sengled bulbs a tubeszb router and an ikea router, the bulbs were rock solid on that network. This was some of the evidence use to conclude it some mix of devices at the time it was only my main network and one customer who had this issue. 

The FW builds I create are based on Koenkk's the difference is only a change in the manufacturer and model name, and the enabling an led to reflect the power state and that the fw is running as it is what controls the led.

stevewull, 2023-02-14:
Thanks. I'll look into it..may be a little while though...busy work week.

stevewull, 2023-02-25:
Update:  I believe the problem is in the cc2652 Router Firmware.

With 94 devices on my Zigbee network, including the 30 Sengled bulbs of various types. I ran the following test:

Unplugged all four of my cc2652 routers. Thus, my mesh network is comprised of a cc2625 controller and five repeaters (4 Sonoff S31 plugs, and a mains CO2 sensor).  For bulbs that had fallen off, I performed a reset/re-pair (turn on/off 10 times). I turned off "Legacy availability payload." I set the transmit power to 20 

Network has been solid. Sengled bulbs stayed online. (With the exception of one lamp that has a four position switch that Sengled does not seem to like.)

Then I added the 4 routers back to the network. Some bulbs soon dropped offline.

I have not yet had a chance to install the 20221102 firmware on the routers. The changelog includes: "Fix directly connected Xiaomi devices disconnecting" which might help. I suspect a similar fix will be need to be made for Sengled bulbs.

tube0013, 2023-02-26:
Thanks for the details. This mirrors the experience of 1 other customer.

As I mentioned before I did some sniffs as well as sent the previous customer a temporary radio that he could compile some sniffs of his network. The Sengled bulbs would experience NWK conflicts for some reason, and the first few times they would correctly respond to the Network's Leave and ReJoin request, however after a some amount of time the network would send a leave-rejoin request and the bulbs just would never respond/come back. 

Obviously there is some interplay with the z-stack routers and the network that in some cases seems to surface this (in my opinion) FW issue with the Sengled bulbs. My opinon is based on it not happening to any other devices on the network mesh. Secondly it seems that only a certain mix of devices on the network cause this issue. First is the shear number of users using z-stack based routers the ones I sell and other who flash the z-stack fw to devices themselves, and the low number reporting issues like this. Second, when I did the investigation into this before I set up a network with a cc2652 coordinator (with no antenna) joined a cc2652 router then joined 2 Sengled bulbs, I believe I added a few more sensors as well. This network and bulbs was rock solid for multiple weeks before I took it down. 

If you are up for trying the 20221102 z-stack FW i'd be interested to hear if it makes a difference. Otherwise reach out to me at store @ tubeszb. com and we can explore options if you want to return them.

tube0013, 2023-03-09:
following up on this. I've compiled an alternate firmware based on the [pvto diy firmware](https://ptvo.info/). 

might be worth trying as well. It's a closed source project but based on change log and assumptions I think it still uses the TI-Z-Stack SDK, so if this is related to z-stack bugs the issue may presist, but the fw should be different enough to test to see if it makes a difference with the connectivity of the Sengleds.

It's flashed the same way

[tubeszb_router_ptvo.zip](https://github.com/tube0013/tube_gateways/files/10933637/tubeszb_router_ptvo.zip)