tube0013/tube_gateways #133: coordinator not responding to zigbee2mqtt?

chrisdrackett, 2023-02-17:
I'm running into an issue where after a couple of days zigbee2mqtt will crash and I'll get the following when it tries to startup:

```
Zigbee2MQTT:error 2023-02-16 16:23:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-16 16:23:51: Failed to start zigbee
Zigbee2MQTT:error 2023-02-16 16:23:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-16 16:23:51: Exiting...
Zigbee2MQTT:error 2023-02-16 16:23:51: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:198:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

I thought I was having the same issue I had before where my coordinator was crashing and was no longer reachable, but when I ping it or go to its IP in the browser it appears to be up and running. That being said, when I restart Z2M I don't get any logs on the coordinator webpage log section until I restart the device and then restart Z2M. At this point I get the output that I would expect:

```
16:28:35	[D]	[streamserver:130]	New client connected from 192.168.1.190
16:28:35	[D]	[binary_sensor:036] 'TubesZB Serial Connected': Sending state ON
```

Not sure what is going on, or how to debug further!

chrisdrackett, 2023-02-17:
probably going to try and roll back the CC2652 Firmware this weekend as that is my hunch at the moment.

ptr727, 2023-02-26:
I just returned home after a week on holiday and my z2m was not working because my coordinator is no longer online.  
This has happened before and I updated the firmware to `20221226`, but seems it is still an issue.  
A power cycle brought it back online again.  
Something not quite right with the network stack.

Currently running running `CC1352P2_CC2652P_launchpad_coordinator_20221226`