tube0013/tube_gateways #140: Update TubesZB CC2652 PoE Coordiniator OTA Network Upgrade Procedure

mrosun, 2023-03-12:
The upgrade procedures [here](https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-poe-2022) over the network, are not up to date.

My coordinator's GUI looks different than what is on the instructions page and I'm not sure which toggle(s) to activate prior to OTA update.

[image]

For example, I don't know which is the toggle "Trigger Zigbee Module Bootloader" or if something has changed.

Also, if we're updating through the web interface, how do we run "Run the cc2538-bsl flasher" command, since there's no text input on the interface?

tube0013, 2023-03-12:
In the Web Interface OTA Update Section is for ESPHome/ESP32 firmware.

For CC2652 - you toggle the Prep the cc2652p2 for firmware update

then on computer in a terminal/command prompt (not the web site) you run the cc-2538-bsl.py script to update the 
firmware. 

Recommend grabbing latest ESPHome firmware from model's fw directory and flashing the new esp32 firmware before doing the cc2652.

mrosun, 2023-03-12:
Ahhh, now I understand, thank you!

tube0013, 2023-03-12:
the screen shots are from newer ESPHome build, the language changed a bit to try to make clearer. sorry for the confusion.

mrosun, 2023-03-12:
I just updated the ESPHome firmware and can see the new interface.

mrosun, 2023-03-12:
When I try to update the cc2652 I get the following error:

`Opening port socket://192.168.47.11:6638, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Firmware file: Intel Hex
Connecting to target...
ERROR: No response from target on status request. (Did you disable the bootloader?)
`

I did follow the instructions to Trigger Zigbee Module Bootloader, as can be seen below:

[image]

Do you have any further advice?

mrosun, 2023-03-12:
Ahh the failed attempts looks like it broke the communication to Home Assistant:

`Zigbee2MQTT:error 2023-03-12 13:53:04: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)`

tube0013, 2023-03-12:
only one thing can access the coordinator at a time, so shut down z2m when doing the update.

https://github.com/tube0013/tube_gateways/blob/main/Troubleshooting.md#zigbee2mqtt-startup---error-srsp---sys---ping-after-6000ms

mrosun, 2023-03-12:
That was the issue! I can't believe I didn't think of that =(

Thanks for everything.

closed 2023-03-13 by tube0013