tube0013/tube_gateways #143: password ?

toxic0berliner, 2023-03-19:
Hello, just got my tubeszb-cc2652-poe-2022 in the mail recently, took me a while to understand I needed to change the panID and network key to be able to start Z2M and finally re-pair each of my 50+ devices... But I love it now it's stable, POE, even fixed some strange issue I had with CP03 devices randomly showing offline...

But I'm a bit baffeled by the fact it's so very wide open... I mean, I got several networks and an opnsense firewall in the middle so I can (and will) block my kids from accessing the device, but hey, I still find it very strange that there seems to be nothing to setup a password to access the dashboard at least, the one that allows flashing and restarting... 

Maybe even have a whitelist of allowed IPs to connect to the serial-over-ip port would be sensible to me...

Is there something I missed or is this just not a feature yet ? Is it on the table for later maybe or am I the only one with this need ?

tube0013, 2023-03-19:
The web ui is a function of esphome. It can be disabled in the fw yaml.

Another option is to to use your firewall/router to limit traffic only to and from the poe device to the z2m or Home Assistant IP.

toxic0berliner, 2023-03-21:
On my firewall it'll only be able to filter traffic from other networks, inside my IOT network this device remains wide open sadly... That's why I'm trying to secure it a bit. 
I fully understand it relies on components outside your contribution, but could you help me a bit at least understand what could be done. 
Dashboards of esphome I think I found what to add in the umm file but how would I get the file onto the device? Should I build a full firmware bin file including my custom umm and then use the GUI to switch to this firmware I custom build?

Also what are the components providing the serial over TCP so I could look into telling them to only listen to packets from my home assistant IP for example.

Thanks in advance for any help.

tube0013, 2023-03-21:
The serial side is currently using an external component I forked and had updated when the original external competent was no longer being updated and no sign of continued development. It since has gotten an update. The links are in the source yaml files posted in this repo.

Adding security got the serial stream side would break all functionality as the mechanisms in z2m and zha which allow connecting to the remote serial port have no ability to anything else with the port. 

Are you concerned your kids will manipulate the tcp serial stream?

I don't know the specifics of your setup but I'd think limiting traffic to just be between 2 devices should be possible on solutions that support vlans. Or just block traffic from one vlan to another.

toxic0berliner, 2023-06-19:
@tube0013 I just discovered recently the ESPHome addon in home assistant and it shows me my tubeszb-cc2652-poe-2022 as discovered and I can "adopt" it. 
Thing is I already have z2m setup with the coordinator using it's IP, and quite a good number of devices connected already...
I don't really know what adopting it might bring besides maybe to centralized esphome page to perform upgrades, but I'd have 2 quick questions : 
- would it reset the coordinator and force me to re-pair everything if I adopt it in esphome ?
- in the GUI, there is a "Zigbee Module Reset" option with an "Action" looking like a checkbox... If I hit it, this will reset the zigbee and it will have forgotten all devices connected to it ?
Thanks in advance for your kind help undestanding what I have setup (and potentially how fragile it all is :D)

tube0013, 2023-06-19:
The ESPHome side of the device has no effect on the zigbee side - it provides only the serial over tcp link. Adopting it would allow for esphome updates which generally I don't recommend as they may break the custom component doing the serial to tcp stream.

There are no actions without other tools that would reset the zigbee module requiring you to re-join everything.

The reset - is a reboot of the zigbee module firmware.