tube0013/tube_gateways #144: Firmware update doesn't work for tube-zb-gw-cc2652p2-v2

Cafun, 2023-04-07:
Hi!

I'd like some help with flashing my tube-zb-gw-cc2652p2-v2 coördinator.
When I run the command bellow, it just returns to the prompt.
[image]

Here is the debug log on the tube:
[image]

I don't know what I'm doing wrong.
Hope you can help. Thanks.

Regards,
Cafun

verstrepeng, 2023-04-07:
Hi,
I think it's because you're missing
`-evw`
in the command line.
KR,
Geert

Cafun, 2023-04-07:
Right! Thanks. It worked.

The [documentation](https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#network) doesn't states the -evw in the command line. I tried it, but got an error the IntelHex library could not be imported. It also isn't in the documentation, so I thought I would not need those options and the IntelHex library.

closed 2023-04-07 by Cafun

tube0013, 2023-04-07:
thank for calling out the issues in the docs, I've updated to include the -evw switch that was missing and also link to the cc2538-bsl's docs on dependencies.

Cafun, 2023-04-08:
Thanks for updaten the docs. Maybe it was obvious for most, but not for me ;-)