tube0013/tube_gateways #147: [CC2652P2 ETH/USB] Error : Failed to connect to the adapter

shad-lp, 2023-05-24:
Hello,

For some weeks now I'm having problems with my CC2652P2 ETH/USB controller.
The serial connection goes off some minutes after I restart the device :

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-05-24 21:39:49: Logging to console and directory: '/app/data/log/2023-05-24.21-39-49' filename: log.txt
Zigbee2MQTT:info  2023-05-24 21:39:49: Starting Zigbee2MQTT version 1.30.3 (commit #24c6b2e)
Zigbee2MQTT:info  2023-05-24 21:39:49: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:info  2023-05-24 21:39:53: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-05-24 21:39:53: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2023-05-24 21:39:53: Currently 2 devices are joined:
Zigbee2MQTT:info  2023-05-24 21:39:53: AQ-SDB (0x00158d0008c11fd8): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-05-24 21:39:53: AQ-CHP (0x00158d000910b4ed): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:warn  2023-05-24 21:39:53: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-05-24 21:39:53: Allowing new devices to join.
Zigbee2MQTT:warn  2023-05-24 21:39:53: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-05-24 21:39:53: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-05-24 21:39:53: Connecting to MQTT server at mqtt://ida.aelnas.ovh:1883
Zigbee2MQTT:info  2023-05-24 21:39:53: Connected to MQTT server
Zigbee2MQTT:info  2023-05-24 21:39:53: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-05-24 21:39:53: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-05-24 21:39:54: MQTT publish: topic 'zigbee2mqtt/AQ-SDB', payload '{"battery":97,"humidity":75.53,"linkquality":65,"power_outage_count":7,"pressure":1012.9,"temperature":20.6,"voltage":2995}'
Zigbee2MQTT:info  2023-05-24 21:39:54: MQTT publish: topic 'zigbee2mqtt/AQ-CHP', payload '{"battery":90,"humidity":46.52,"linkquality":112,"power_outage_count":6,"pressure":1012.3,"temperature":20.6,"voltage":2985}'
Zigbee2MQTT:info  2023-05-24 21:39:54: Zigbee2MQTT started!
Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-05-24 21:43:24: Logging to console and directory: '/app/data/log/2023-05-24.21-43-24' filename: log.txt
Zigbee2MQTT:info  2023-05-24 21:43:24: Starting Zigbee2MQTT version 1.30.3 (commit #24c6b2e)
Zigbee2MQTT:info  2023-05-24 21:43:24: Starting zigbee-herdsman (0.14.103)
Zigbee2MQTT:error 2023-05-24 21:43:44: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-05-24 21:43:44: Failed to start zigbee
Zigbee2MQTT:error 2023-05-24 21:43:44: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-05-24 21:43:44: Exiting...
Zigbee2MQTT:error 2023-05-24 21:43:44: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

And so on, looping indefinitely.

In the mean time, on the TubeZB interface :

```
21:50:21 | [I] | [restart.button:012] | Restarting device...
-- | -- | -- | --
21:50:21 | [I] | [restart.button:012] | Restarting device...
21:50:21 | [I] | [app:134] | Rebooting safely...
21:51:47 | [D] | [streamserver:102] | Client 192.168.100.222 disconnected
21:51:47 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
21:51:52 | [D] | [streamserver:074] | New client connected from 192.168.100.222
21:51:52 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
21:52:11 | [D] | [streamserver:102] | Client 192.168.100.222 disconnected
21:52:11 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
21:52:17 | [D] | [streamserver:074] | New client connected from 192.168.100.222
21:52:17 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
```

Sometimes it stays online for a while, but definitely can't be reached from z2m few minutes later.

Everything was running smooth for the last months.
I have not yet upgraded the firmware (can't find the version I'm using right now, where should I check ?).
I have downgraded z2m version to 1.30.2, but same behaviour, so I guess it's not related to z2m.

Thanks in advance.

tube0013, 2023-05-24:
please see the troubleshooting guide here:

https://github.com/tube0013/tube_gateways/blob/main/Troubleshooting.md

shad-lp, 2023-05-24:
Thanks for the quick answer :

- I had to install pyserial and intelhex package from pip3
- Run the script :

```
./cc2538-bsl.py -p socket://tubezb.xxxxxx.ovh:6638 -evw ./CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Opening port socket://tubezb.xxxxxx.ovh:6638, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_coordinator_20221226.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:2A:2E:C8:C6
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xa9dc145d)
```

After 2 minutes, I rebooted the device and started zigbee2mqtt again.
This time, no more state on/off loop.

So far, so good, z2m started successfully :

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-05-24 22:33:36: Logging to console and directory: '/app/data/log/2023-05-24.22-33-36' filename: log.txt
Zigbee2MQTT:info  2023-05-24 22:33:36: Starting Zigbee2MQTT version 1.30.4 (commit #b2dd21e)
Zigbee2MQTT:info  2023-05-24 22:33:36: Starting zigbee-herdsman (0.14.111)
Zigbee2MQTT:info  2023-05-24 22:36:40: zigbee-herdsman started (restored)
Zigbee2MQTT:info  2023-05-24 22:36:40: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20221226,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2023-05-24 22:36:40: Currently 2 devices are joined:
Zigbee2MQTT:info  2023-05-24 22:36:40: AQ-SDB (0x00158d0008c11fd8): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2023-05-24 22:36:40: AQ-CHP (0x00158d000910b4ed): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:warn  2023-05-24 22:36:40: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2023-05-24 22:36:40: Allowing new devices to join.
Zigbee2MQTT:warn  2023-05-24 22:36:40: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2023-05-24 22:36:40: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-05-24 22:36:40: Connecting to MQTT server at mqtt://ida.aelnas.ovh:1883
Zigbee2MQTT:info  2023-05-24 22:36:41: Connected to MQTT server
Zigbee2MQTT:info  2023-05-24 22:36:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-05-24 22:36:41: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2023-05-24 22:36:41: MQTT publish: topic 'zigbee2mqtt/AQ-SDB', payload '{"battery":97,"humidity":75.53,"linkquality":65,"power_outage_count":7,"pressure":1012.9,"temperature":20.6,"voltage":2995}'
Zigbee2MQTT:info  2023-05-24 22:36:41: MQTT publish: topic 'zigbee2mqtt/AQ-CHP', payload '{"battery":90,"humidity":46.52,"linkquality":112,"power_outage_count":6,"pressure":1012.3,"temperature":20.6,"voltage":2985}'
Zigbee2MQTT:info  2023-05-24 22:36:41: Zigbee2MQTT started!
```

I'll monitor the logs for the next days, thanks again for the prompt answer.

closed 2023-05-24 by shad-lp