tube0013/tube_gateways #150: Update firmware to match koenkk 20221102

apadua, 2023-07-06:
Curently available firmware version is based on older release of the Stock Firmware. Scalability issues were addressed in the newest firmware. Is there an ETA on when TubesZB router firmware will have parity with Koenkk's native fw?

tube0013, 2023-07-06:
Recommend updating to Koenkk’s if you want this now. I will look to update in coming weeks as time allows.

apadua, 2023-07-06:
Thank you very much for the response. I can wait.

tube0013, 2023-08-25:
I pushed a new router version bringing it up to the z-stack 1102 version. I only modified the led to be always enabled so there is an indicator of power and fw boot. 

The descriptors for model and manufacturer are unchanged meaning they are the same as used in the z-stack repo version so I do not need to maintain a converter or quirk. TubesZB is coded in the product label attribute under the basic cluster.

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-router/firmware