tube0013/tube_gateways #153: [SUGGESTION] ESP32-C6 (or ESP32-H2) to Ethernet gateway/bridge hardware product?

Hedda, 2023-08-16:
Like to suggest that you consider looking into designing an ESP32-C6 (or ESP32-H2) to Ethernet gateway/bridge hardware product.

While it is still early days for ESP32-C6 and ESP32-H2 since both those SoCs contain an IEEE 802.15.4 radio with Zigbee and Thread support they could be an alternative to Texas Instruments and Silicon Labs based SoCs, which should at the very least make such a  Ethernet gateway/bridge very interested to developers of DIY firmware like Tasmota and ESPHome.

FYI, grobasoz added some Zigbee RCP and CLI firmware images for ESP32-C6 (no info on what he tested with those so far):

* https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6
  * https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/ZigbeeCLI
  * https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/Zigbee_RCP

A bridge product that could use such firmware for also be interesting as a remote Zigbee Coordinator adapter with serial stream proxy server for network-attached bridging to Zigbee gateway solutions based in zigpy, like [ZHA integration component for Home Assistant](https://www.home-assistant.io/integrations/zha/) and the [Zigbee Plugin for Domoticz](https://www.domoticz.com/wiki/ZigbeeForDomoticz), as DamKast recently announced "zigpy-zboss" as an (unofficial and experimental) ZBOSS radio library for zigpy (https://github.com/kardia-as/zigpy-zboss) which should hopefully make zigpy based solutions compatible with the Espressif's ESP ZBOSS Zigbee stack that those the libraries for ESP32-C6 and ESP32-H2 are based on.   

Note that scenario for zigpy based solution hinges on if the CLI in Espressif's Espressif's ESP ZBOSS Zigbee stack (esp-zboss-lib from Espressif) is included in the ESP32-C6/ESP32-H6 firmware and if they use the standard ZBOSS API/CLI which if it is the case should make it usable as a remote network-attached Zigbee Coordinator for Home Assistant's ZHA integration if configure via serial server pass-through as proxy using a network socket -> https://github.com/zigpy/zigpy/issues/1052 (More zigpy-zboss discussion related to zigpy here -> https://github.com/zigpy/zigpy/issues/394)

CSA (Connectivity Standards Alliance) which governs the new Matter standard l have made it clear that bridges for protocol-XYZ to Matter will play an extremely important role in the adoption of Matter as the default norm as many people will still want to keep using their old devices but bridge them to a Matter fabric, so I believe that both ESP32-C6 or ESP32-H) with Ethernet PHY for wired LAN interface could be very interesting for many to make Zigbee-to-Ethernet and Thread-to-Ethernet bridge/gateway solutions for stable connections to a user's network, in the same spirit as your existing Zigbee to Ethernet hardware products based on ESP32.

An alternative to designing/making a one-SoC board that could possibly only be used as a remote network-attached Zigbee Coordinator adapter with just a serial stream proxy server bridge could be to make a board that instead uses two ESP32 SoCs such as Espressif's own example reference solution then it should be possible to use them a a full-blown Zigbee-to-Matter/MQTT or Thread-to-Matter/MQTT gateway solution.

That concept would as well be in line with Tasmota's existing Zigbee2Tasmota (Z2T) support -> https://tasmota.github.io/docs/Zigbee/

Check out Espressif's new ESP Thread Border Router / Zigbee Gateway board with optional Ethernet Sub board from as a reference:

https://www.cnx-software.com/2023/06/20/espressif-esp-thread-border-router-board-combines-esp32-h2-esp32-c3-wireless-chips/

https://www.hackster.io/news/espressif-launches-a-kit-to-turn-its-esp32-modules-into-certified-thread-border-routers-for-matter-c61db0fce050

https://github.com/espressif/esp-thread-br

https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/

https://www.aliexpress.us/item/1005005688193617.html

For reference, an earlier two-board ESP Thread Border Router solution was previously available as an example gateway/bridge project:

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br

https://openthread.io/guides/border-router/espressif-esp32

Such a two-SoC setup might initially be more flexible and useful in a proof-of-concept as a remote network-attached Zigbee Coordinator where an ESP32-S3 would be flashed with for example ESPHome or Tasmota and then have ESP32-C6 only run Zigbee/OpenThread firmware to simply act as a Zigbee/Thread radio connected via UART or SPI using CLI or RCP interface for Zigbee or OpenThread.

PS: I read that Tasmota already implemented official initial support of the Matter standard since Tasmota v13 and its developers are currently looking into implementing bridge mode for Matter, and though development on their Thread support for ESP32-C6 and ESP32 have not yet started it is on their roadmap, see https://tasmota.github.io/docs/Matter/ (and https://github.com/arendst/Tasmota/discussions/17872)

michcfr, 2024-02-11:
Hello,
Any update about this new design?
thank you

tube0013, 2024-02-11:
None on my end. Espressif seems to be working on a python library compatible with zigpy. So will watch that. Not promising anything.

Hedda, 2024-02-11:
> Espressif seems to be working on a python library compatible with zigpy. So will watch that.

Yes an Espressif developer has now posted an experimental version of zigpy-espzb radio library for zigpy (but there is no prebuilt firmware images and the library is not yet available on PyPI so not user friendly to test for non developers):

https://github.com/lhespress/zigpy-espzb

https://github.com/zigpy/zigpy-cli/pull/45