tube0013/tube_gateways #154: tubeszb-cc2652-poe-2022 Crashing

bsaurusrex, 2023-08-25:
Hello,  I have been deploying multiple ZY-M100-L tuya human presence sensors (in ceiling version).  I am noticing they are super chatty, sending messages every second.

The problem is, these chatty devices seem to be overloading and crashing the POE cc2652 once a day minimally with error 
`error 2023-08-24 13:40:50: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`
. 
I am wondering if there are any logs that I can share to help make the coordinator sensor more stable?

This is the log that keeps getting spammed
```

nfo  2023-08-25 08:57:46: MQTT publish: topic 'zigbee2mqtt/OF_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"OF_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c1387664b2f9f9","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":27011,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":61.5,"illuminance_lux":14,"last_seen":"2023-08-25T08:57:46+08:00","linkquality":83,"maximum_range":3.45,"minimum_range":0.75,"presence":true,"radar_sensitivity":6,"target_distance":3.44}'
debug 2023-08-25 08:57:46: Received Zigbee message from 'KT_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,154],"type":"Buffer"},"datatype":2,"dp":109}],"seq":29952}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:46: MQTT publish: topic 'zigbee2mqtt/KT_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"KT_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c138d1e44fe112","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":31773,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":5,"illuminance_lux":7,"last_seen":"2023-08-25T08:57:46+08:00","linkquality":134,"maximum_range":6,"minimum_range":0.6,"presence":true,"radar_sensitivity":7,"target_distance":4.1}'
debug 2023-08-25 08:57:46: Received Zigbee message from 'KT_presence_2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,225],"type":"Buffer"},"datatype":2,"dp":9}],"seq":65280}' from endpoint 1 with groupID 0
debug 2023-08-25 08:57:46: Received Zigbee message from 'MBB_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":109}],"seq":32000}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:46: MQTT publish: topic 'zigbee2mqtt/KT_presence_2', payload '{"cli":" ","detection_delay":0.1,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"KT_presence_2","hardwareVersion":1,"ieeeAddr":"0x385b44fffe149578","manufacturerID":4098,"manufacturerName":"_TZE204_ztc6ggyl","model":"ZY-M100-L","networkAddress":57613,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":60,"illuminance_lux":67,"last_seen":"2023-08-25T08:57:46+08:00","linkquality":47,"maximum_range":5,"minimum_range":0.6,"presence":true,"radar_sensitivity":7,"self_test":"check_success","target_distance":2.25}'
info  2023-08-25 08:57:46: MQTT publish: topic 'zigbee2mqtt/MBB_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"MBB_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c138b46a593ccc","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":2899,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":15,"illuminance_lux":19,"last_seen":"2023-08-25T08:57:46+08:00","linkquality":94,"maximum_range":3.75,"minimum_range":0.6,"presence":false,"radar_sensitivity":5,"target_distance":0}'
debug 2023-08-25 08:57:47: Received Zigbee message from 'MBB_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":109}],"seq":32000}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:47: MQTT publish: topic 'zigbee2mqtt/MBB_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"MBB_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c138b46a593ccc","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":2899,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":15,"illuminance_lux":19,"last_seen":"2023-08-25T08:57:47+08:00","linkquality":94,"maximum_range":3.75,"minimum_range":0.6,"presence":false,"radar_sensitivity":5,"target_distance":0}'
debug 2023-08-25 08:57:47: Received Zigbee message from 'JR_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":109}],"seq":52992}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:47: MQTT publish: topic 'zigbee2mqtt/JR_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"JR_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c138c4f9941f9e","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":18871,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":57.5,"illuminance_lux":8,"last_seen":"2023-08-25T08:57:47+08:00","linkquality":69,"maximum_range":3,"minimum_range":1.65,"presence":false,"radar_sensitivity":6,"target_distance":0}'
debug 2023-08-25 08:57:47: Received Zigbee message from 'KT_presence_2', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,69],"type":"Buffer"},"datatype":2,"dp":9}],"seq":0}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:47: MQTT publish: topic 'zigbee2mqtt/KT_presence_2', payload '{"cli":" ","detection_delay":0.1,"device":{"applicationVersion":70,"dateCode":"","friendlyName":"KT_presence_2","hardwareVersion":1,"ieeeAddr":"0x385b44fffe149578","manufacturerID":4098,"manufacturerName":"_TZE204_ztc6ggyl","model":"ZY-M100-L","networkAddress":57613,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":60,"illuminance_lux":67,"last_seen":"2023-08-25T08:57:47+08:00","linkquality":51,"maximum_range":5,"minimum_range":0.6,"presence":true,"radar_sensitivity":7,"self_test":"check_success","target_distance":3.25}'
debug 2023-08-25 08:57:47: Received Zigbee message from 'OF_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,89],"type":"Buffer"},"datatype":2,"dp":109}],"seq":25600}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:47: MQTT publish: topic 'zigbee2mqtt/OF_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"OF_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c1387664b2f9f9","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":27011,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":61.5,"illuminance_lux":14,"last_seen":"2023-08-25T08:57:47+08:00","linkquality":83,"maximum_range":3.45,"minimum_range":0.75,"presence":true,"radar_sensitivity":6,"target_distance":3.45}'
debug 2023-08-25 08:57:48: Received Zigbee message from 'KT_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,185],"type":"Buffer"},"datatype":2,"dp":109}],"seq":30208}' from endpoint 1 with groupID 0
info  2023-08-25 08:57:48: MQTT publish: topic 'zigbee2mqtt/KT_presence_1', payload '{"detection_delay":0.1,"device":{"applicationVersion":74,"dateCode":"","friendlyName":"KT_presence_1","hardwareVersion":1,"ieeeAddr":"0xa4c138d1e44fe112","manufacturerID":4417,"manufacturerName":"_TZE204_sxm7l9xa","model":"ZY-M100-S_1","networkAddress":31773,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"fading_time":5,"illuminance_lux":7,"last_seen":"2023-08-25T08:57:48+08:00","linkquality":134,"maximum_range":6,"minimum_range":0.6,"presence":true,"radar_sensitivity":7,"target_distance":4.41}'
debug 2023-08-25 08:57:48: Received Zigbee message from 'MBB_presence_1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":109}],"seq":32256}' from endpoint 1 with groupID 0
info
```

Any recommendations or other debug logs I can pull?

tube0013, 2023-08-25:
Try reflashing the ESPHome fw over serial as described in the models page. 

If it continues let me know can try a different build.

bsaurusrex, 2023-10-11:
Updated the esphome and the CC2652 firmwares. Still crashing daily unfortunately,

tube0013, 2023-10-11:
Yeah those send a ton of data, I've used them in testing for that reason lol. let me see if I can through together a firmware based on the IDF platform in esphome as it may be a bit more efficient. You'd need to reflash over serial though to take advantage though. I see if I can get to this tonight, if not tomorrow.

tube0013, 2023-10-14:
do you have the 2022 or 2023 version of the PoE 2652 Coordinator? (2023 started shipping in May of this year and will have 2 LEDs on it, 2022 will not).

bsaurusrex, 2023-10-15:
22

tube0013, 2023-10-15:
Can you try this esphome FW - you will need to flash over usb as it is based on the ESP_IDF platform:

[tubeszb-cc2652-poe-idf-2022.bin.zip](https://github.com/tube0013/tube_gateways/files/12908278/tubeszb-cc2652-poe-idf-2022.bin.zip)

bsaurusrex, 2023-10-25:
So far the stability seems to be improved.  Previously, it was crashing every 6-12 hours.  Now it hasn't crashed yet.

Still monitoring for a few more days.

bsaurusrex, 2023-10-26:
Crashed after 32 hours.  Certainly better than every 8 hours with this firmware.

bsaurusrex, 2023-10-26:
[image]

bsaurusrex, 2023-10-26:
Suggestion: is it possible to create an HA addon watchdog service that auto reboots the coordinator?

bsaurusrex, 2023-10-27:
Temporarily, I created a rest_command in configuration.yaml which I can call every 8 hours with an automation.

rest_command:
  reboot_zb_coordinator:
    url: "http://192.168.1.4/button/restart_the_device/press"
    method: POST
    headers:
      Accept: "*/*"
      Accept-Language: "en-GB,en-US;q=0.9,en;q=0.8"
      Connection: "keep-alive"
      Content-Type: "text/plain;charset=UTF-8"
      Origin: "http://192.168.1.4"
      Referer: "http://192.168.1.4/"
      User-Agent: "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/118.0.0.0 Safari/537.36"
    verify_ssl: false

zigbee2mqtt seems to handle this gracefully.

tube0013, 2023-10-27:
I'm still not sure why this is occurring so frequently for you unless it's a hardware issue at this point. 

Can you reach out to store@tubeszb.com as I'd like to try and send you a replacement esp32-poe to see if it solves this for you.

LordNex, 2023-10-27:
What version is everyone running? When i adopt it into ESP Home 10.0.3 it obviously wants an update. But if you look at the yaml code that came from the adoption it points to Tube's distro yaml file. Should we go ahead and let ESPHome update it at some point? I've let some ESP8266 devices go for over a year without updating and by time I tried the version was in correct, wouldn't adopt, and I basically had to reverse engineer the sensor I made. This looks like the one I'm using, I can tell by the heartbeat symbol which usually isn't there after and ESPHome Update Push.

bsaurusrex, 2023-10-27:
@tube0013 The frequency is because I have a lot of tuya presence sensors (x11) with varying degrees of spamming.  I think I am just super stress testing it.

LordNex, 2023-10-27:
> @tube0013 The frequency is because I have a lot of tuya presence sensors (x11) with varying degrees of spamming. I think I am just super stress testing it.
> 
> Do you think its worth replacing or is a better path to give up on these presence sensors.

I have about 18 devices and growing. I'm trying to replace all my cabinet and closet lights with ZigBee's running on main power, so that they can act as a repeater. so far about 6 mains connected devices with the rest various Aqura window\door sensors, water sensors, water shut off valve, and several buttons and temp sensors. When I was using a Nortek HUBZ1 i couldn't keep anything that wasn't more than 3 feet away from a mains connected device to stay in the mesh. I could easlly repair it and it would even keeps its name, but having to do that every other day is consider home automation working. We've gotten so reliant on Zigbee that if it isnt functioning right, the house doesn't. Luckily all my wall switches and wall outlets are TP Link so they work pretty good as long as you keep them local.

Im wondering is is posiable or even preferable to have several different coordinators on a network? I changed my wifi to only run 80 mhz channels on the 5ghz radio, moved everything over to WPA3, and set 20/40mhz coexistence support on my wifi 6 AP. I have Aruba IAP 335's and a IAP 275 Outdoor running in campus mode so they form thier own self healing network. But i wanted to get my iPhones of of those APs so I put in a Netgear wifi 6 AP in my sons room and set it to Wifi 6 and the same SSID as the primary network, then trunked a line from my Aruba 2930 Managed Switch. Needless to say I have tones of 2.4 ghz around me but i live in the country so my neighbors are not right up my ass with theirs APs

midget214, 2023-11-17:
I've started to see a similar issue with my 2022 PoE adapter.  In my case I have a single motion sensor connected to it.  Its been working flawlessly until about 2 weeks ago.  Nothing has changed on the firmware side.  I will try to reflash the firmware once I find some time over the weekend.