tube0013/tube_gateways #155: Firmware update question

Bloodpack, 2023-09-08:
Hello i recieved my `tubeszb-cc2652-poe-2023 coordinator` and i want to update the firmware of the `CC2652P` and `Olimex ESP-Poe`

i installed the Homeassistant Addon `TubesZB TI CC2652 FW Flasher` 
my coordinator shows FW version: `20221226` and the newest FW is [20230507](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip)
can i update directly without breaking anything?

How can i see what FW version of ESP-home my coordinator is running?

Can i adopt the coordinator to my ESP-home and use this [yaml](https://raw.githubusercontent.com/tube0013/tube_gateways/main/models/current/tubeszb-cc2652-poe-2023/firmware/esphome/tubeszb-cc2652-poe-2023.yaml) from your repo so i can have it updated from my ESP-home

sorry for the maybe dumb questions but i want to make sure i don´t break anything

tube0013, 2023-09-08:
The Z-Stack 0507 firmware has not been the smoothest release. Which is why I have not shipped devices with it. There is a long thread related to it I. The firmware repo. So I don't currently recommend it, I suspect a new release will be out in the next month or 2 and hopefully prove to be more stable. Having said that I can't prevent anyone from updating if they choose. 

The ESPHome side relies on a custom component, updating the device to the newest ESPHome release is not recommended as I'm not able to test and confirm it's working reliably with every release. You can always fall back to the binary on GitHub which is what is flashed on devices when shipped.

Bloodpack, 2023-09-08:
Thank you for your fast response....

i will stay on the current installed firmware, for now......

is there a way to show the currently running version of ESPHome on my coordinator?

tube0013, 2023-09-08:
I believe currently the only place to see it is via the ESPHome log.

closed 2023-09-08 by Bloodpack