tube0013/tube_gateways #156: [REQUEST] MGM240P (e.i. Silicon Labs EFR32MG24) based Zigbee Coordinator and Zigbee Router adapters

Hedda, 2023-09-12:
@tube0013  Please consider making/selling MGM240P (Silicon Labs EFR32MG24) based Zigbee Coordinator and Zigbee Router adapters:

https://www.silabs.com/wireless/zigbee/efr32mg24-series-2-modules

https://fccid.io/QOQ-GM240P

https://fcc.report/FCC-ID/QOQ-GM240P/

https://www.silabs.com/documents/public/data-sheets/mgm240p-datasheet.pdf

Suggesting updated revisions of your existing "EFR32 MGM21 Ethernet/USB Serial Coordinator" and "EFR32 MGM21 PoE Coordinator":

https://tubeszb.com/product/efr32-mgm21-ethernetusb-serial-coordinator/

https://tubeszb.com/product/efr32-mgm21-poe-coordinator/

I think that there are probably a lot of people in the Home Assistant community who would jump at buying such cutting-edge devices.

The main benefit and major selling point with the EFR32MG24 SoC is that the MGM240 modules for Zigbee is that it has higher RAM and larger flash than the older EFR32MG21/MGM210P modules  (256kB RAM + 1536kB Flash instead of only 96kB RAM + 1024kB Flash) so should, at least in theory, be able to handle more than twice the total amount of Zigbee 3.0 devices even at the highest security levels.

EFR32MG24/MGM240 is also said to be better at handling concurrent Zigbee NCP and Thread RCP so that it can be simultaneously used as a Zigbee Coordinator and Thread Border Router for the Matter protocol standard at the same time, again at least in theory.

This will be great if and when it could also be used as a Thread Border Router too as you can have more than Thread Border Routers.

Also see related zigpy (ZHA) discussion:  

https://github.com/zigpy/zigpy/discussions/893

PS: Believe MGM240P parts should be available in quantity by now yet I can still not find anyone else making MGM240P or EFR32MG24 based Zigbee Coordinator or Zigbee Router USB/Ethernet adapters, so if you are fast then you could be first to market with such devices.

Hedda, 2023-09-12:
FYI, upcoming EFR32MG27 and EFR32MG28 (from the XG27 and XG28 series) is said to be released next year or later this year, but it will likely take even longer before any MGM270/MGM270P or MGM280/MGM280P modules are available, Those are rumoured to support concurrent multi-band/dual-band 2.4 GHz + Sub-GHz radio operation for long range capabilities.

Zigbee Sub-GHz (sub-1 GHz) for smart home automation use is a long time coming (though the technology has already existed for many years and is today commonly used for utilities Zigbee Smart Energy (ZSE / Zigbee SE) metering in many commercial meters. The biggest thing missing has been multi-band/dual-band and dual-mode support in a single chip that is as inexpensive as the others.

https://csa-iot.org/newsroom/subghz-reliable-mesh/

https://rethinkresearch.biz/articles/zigbee-gets-dual-mode-2-4ghz-sub-ghz-spec-longer-range-mesh/

Hedda, 2023-09-12:
Btw, Texas Instruments also released C2674P (CC2674P10) as well as CC1354P (CC1354P10) Sub-1 GHz and 2.4 GHz SoCs.

https://www.ti.com/product/CC2674P10

https://www.ti.com/product/CC1354P10

Hedda, 2024-03-04:
FYI, RFStar has now released Silicon Labs EFR32MG24 based radio modules called "RF-BM-MG24B2" and "RF-BM-MG24B1":

https://www.rfstariot.com/rf-bm-mg24b2-matter-zigbee-openthread-ble-multiprotocol-efr32mg24-module_p106.html

https://www.rfstariot.com/rf-bm-mg24b1-ble5-3-matter-zigbee-thread-efr32mg24-module_p105.html

tube0013, 2024-03-04:
Yes, I've had some for a while in testing they work well, but only are available with a pcb antenna. This is hard to market I think, despite my testing showing only around a 1-2 dbi drop compared to a silabs module with external antenna 🤷🏼‍♂️

LordNex, 2024-03-04:
> Yes, I've had some for a while in testing they work well, but only are available with a pcb antenna. This is hard to market I think, despite my testing showing only around a 1-2 dbi drop compared to a silabs module with external antenna 🤷🏼‍♂️

Are you referring to the board controller interface or the actual antenna that attaches?

LordNex, 2024-03-05:
Just for context I took the one you had off for a larger antenna with a magnetic base so I could run it out to a ceiling HVAC return vent. Helped centralize the primary coordinator and seemed to give a better signal. 

I also figured it might be better to also get the transmission farther away from the controller PCB as well so not interference from it occurred. 

Currently I'm in a semi down state. Not because of the coordinator but more because Home Assistant started acting ,up along with a few other systems, and I feel it's time for a reload and rebuild. 

Off Topic a bit, anyone know a quick small all outside of Home Assistant to run ZigBee off of until I get back up with the main build? 

Normally my system runs off a Rack Mounted PowerEdge Server running VMWare ESXi. I have a lot of cats and no matter what I try I constantly have small hair buildup and it's time I take the server apart and blow her out and reinstall everything. I do have either a RP4 8gig or a Nvidia Jetson Nano as well I could use temporarily. 

I'd put HA on one of those but my build usually runs 16 gigs easily with everything.

Hedda, 2024-03-05:
> Are you referring to the board controller interface or the actual antenna that attaches?

RF radio modules usually come in different models with either [a U.FL/I-PEX mini RF connector for external antenna](https://en.wikipedia.org/wiki/Hirose_U.FL) or with some kind of built-in antenna (such as a [PBC trace circuit board antenna or a soldered ceramic chip antenna](https://www.wellpcb.com/chip-antenna-vs-pcb-antenna.html)), or a combination of two to allow for chocie where one is disabled when the other is in use. See examples:

[image]

[image]

[image]

[image]

LordNex, 2024-03-05:
I would push more for the U.FL than a trace board for anything besides the coordinator. Not unless you made a whole separate PCB as trace I would worry about crosstalk or interference from the overall device. Being as these are the center of our spoke, even though mesh, is still the most sensitive and important member of any good Zigbee designs I've seen.  I like having the choice to go for a higher gain or even directional antenna setup. 

The one thing I still don't like is that central point of failure. Being mesh by design gives it a lot of pathways.  It without the singular coordinator, the network is dead. 

Now if you could build a PCB that would run some form of software that could take multiple coordinators and cluster them, or even have a fall back. Would greatly improve the design and overall interoperability

Hedda, 2024-03-11:
> I've had some for a while in testing they work well, but only are available with a pcb antenna. This is hard to market I think, despite my testing showing only around a 1-2 dbi drop compared to a silabs module with external antenna 🤷🏼‍♂️

Maybe you could consider making a slimline Zigbee USB dongle similar in format to the Home Assistant SkyConnect USB stick?

Perhaps call it something cool like "TubesZB MGM24 Slimline USB Dongle" 😉 

Anyway, believe any EFR32MG24 based adapter made available first would sell just because they will have this very latest chip.

And again, at least in theory it will also be able to control more Zigbee devices with it having lager flash storage and more RAM.