tube0013/tube_gateways #160: Unable to create ZHA Network with EFT32-MGM210-POE-2023

VonJedi, 2023-10-14:
Not sure if I'm doing or done something wrong here.

I plugged in the device with PoE and it's powered up okay, I can get to the web interface, I attempted to set up the ZHA and it seems to hang on the 'Create a Network' page, then just says 'Error' and an okay button.
[image]
[image]

Shows this in the web interface for the device:
[image]

I get this in the error logs of HA.
[image]

Not sure what I'm doing wrong, I updated the firmware on the device using the micro usb, and updated HA to the latest version (2023.11). Still doing it.

Any ideas?

tube0013, 2023-10-14:
Are you using HASOS - with add-ons? if yes can you try using my addon for firmware updates and try re-flashing the zigbee module. 

https://github.com/tube0013/tubeszb_addons/blob/main/tzb-silabs-flasher/DOCS.md

the URL for the firmware file is:

https://github.com/tube0013/tube_gateways/raw/main/models/current/tubeszb-efr32-MGM210-poe/firmware/efr32_MGM210/ncp/7.2.3/MGM21_2022_ncp-uart-7.2.3.gbl

Software flow

after you start the login check and refresh the logs, if it fails, let me know what the logs say.

thanks.

VonJedi, 2023-10-14:
Hey mate,

No errors, seemed to flash fine. 

Still having the same issue though. So strange.

tube0013, 2023-10-14:
Please enable debug logging for zha as per the zha component docs, adding the lines to configuration.yaml. Restart HA, and go through trying to set up zha again. After the failure upload the log - or email to me at store@tubeszb.com hopefully can figure this out.

If the flash worked with no errors, the flashing script was able to speak to the coordinator via ezsp to put it into bootloader mode then successfully send the fw, so I'm hoping the debug logging will give some insight into why zha is failing setup.

Are you doing manual setup up or going through the discovered device flow?

Thanks

revulsion, 2023-10-14:
I am also having this exact problem.  if you still need logs, please let me know

tube0013, 2023-10-14:
Are you setting up manually or via the auto discovered device?

revulsion, 2023-10-14:
Manually.

revulsion, 2023-10-14:
Manual

I am probably making things overly complicated as I have: 
1. HA is running in Kubernetes via the official home-assistant :latest container image 
2. My "iot" devices get their own subnet (routing works and other devices are work, eg an rtsp camera)

tube0013, 2023-10-14:
Can you pull the debug logs when trying to add? 

I'm going to try and replicate shorty in a test environment to see if something is up with current HA

revulsion, 2023-10-14:
[logs-2.txt](https://github.com/tube0013/tube_gateways/files/12908247/logs-2.txt)

tube0013, 2023-10-15:
Okay I was able to replicate this. It seems to be with the latest HA release and backend libs.

as a work around you can roll back to 2023.9.3 and set it up, and then re-update to 2023.10.3, I've confirmed this works.

Working with the ZHA dev to identify the issue.

tube0013, 2023-10-15:
pretty sure it's from this. https://github.com/zigpy/bellows/pull/584

revulsion, 2023-10-15:
Yes, rolling back to 2023.9.3 allowed me to add it.

VonJedi, 2023-10-15:
Yeah that's done it. Perfect. Thanks mate.

tube0013, 2023-11-05:
Confirmed network formation works in the 2023.11.x releases. So closing this.

If anyone comes to this issue and is running 2023.10.x the solution is to downgrade to 2023.9.x  set up ZHA then upgrade back to 2023.10.x or just upgrade to 2023.11.x or greater and all should be okay.

closed 2023-11-05 by tube0013

LordNex, 2023-11-05:
Question, does that mean it's safe to let ESPHome update the coordinator now?  It seems after adoption that it's just a small snippet of code pointing to your yaml file. Just wondering as ESPHome is still at version 10.6 in HA 11.1

tube0013, 2023-11-05:
Please don't comment on closed issues unless it is related to the closed issue. 

You can update the ESPHome fw at your own risk, the fw depends on external components that could break with an update at anytime. You can always roll back with the provided binary provided here.