tube0013/tube_gateways #162: Unable to update firmware on EFR32 MGM21 Ethernet/USB

davidrule1969, 2023-10-28:
I just received a new EFR32 MGM21 Ethernet/USB device.  When I first tried powering it up, I got no LEDs on the ethernet module or the main PCB.  I popped off the cover and noticed that the ethernet module was installed incorrectly, the pins were offset by one.  I don't know if powering the unit up in this state has caused a problem or not.  Anyway, I disconnected the ethernet module and reconnected it in the correct position and I get LEDs and everything seems ok, except I cannot get ZHA or Zigbee2MQTT to recognise it.

Zigbee2MQTT gives me this error:

`Zigbee2MQTT:error 2022-10-14 23:42:17: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)`

ZHA gives a blank dialog box with the title ERROR.

The ESPHome interface is working fine.  However, I refreshed it anyway as some other postings here suggested I do so.  No problems there, ESPHome is still working.

So I thought I'd try updating the firmware on the EFR32.  This seems to have made things even worse!

I followed the YouTube video here: https://www.youtube.com/watch?v=zKrISuWEzL4

After uploading the gbl file (I'm using 20230219_MGM12ncp-uart-sw_7.2.1_2022.gbl) with XMODEM, the serial connection doesn't seem to recognise the end of the upload.  It just sits there at CCCC.

Now when I reset the EFR32, regardless of whether the Bootloader pin is set to Low or High, it always returns to the menu and I never get the '~' to indicate that it is running.

`Gecko Bootloader v2.01.01
1. upload gbl
2. run
3. ebl info
BL >`

Any help gratefully appreciated!
-David

tube0013, 2023-10-28:
Hi, sorry about the mis-aligned esp32 module! 

ZHA has a bug in the current 2023.10.x version for setting up network connected coordinators. It will work in 2023.9.x if you downgrade once set up you can re update to 2023.10.x. Alternatively you can update to the 2023.11 beta and it should work there as well.

For z2m you if you choose to go that way despite the warnings with a efr32 coordinator being experimental. You have to make sure under the port line you tell it ezsp otherwise it assumes a TI adaptor which is why you see the error you got.

davidrule1969, 2023-10-28:
Thanks for the prompt response!

I updated to 2023.11 beta and now ZHA says "Failed to connect".  I suspect this is because I messed up the EFR32 firmware.  How can I recover this?

Connecting via serial I always get the bootloader menu

`Gecko Bootloader v2.01.01`
`  1. upload gbl`
`  2. run`
`  3. ebl info`

I no longer get the ~ prompt regardless of the state of the Bootloader pin.

tube0013, 2023-10-28:
Are you using HASOS with addons?

if so you can use the addon I have here:

https://github.com/tube0013/tubeszb_addons/blob/main/tzb-silabs-flasher/DOCS.md

the Firmware link to use is:
https://github.com/tube0013/tube_gateways/raw/main/models/current/tubeszb-efr32-MGM210-poe/firmware/efr32_MGM210/ncp/7.2.3/MGM21_2022_ncp-uart-7.2.3.gbl

if not using hasos you can download the the firmware linked above and use the universal-siliconlabs-flasher from nabu casa:

https://github.com/NabuCasa/universal-silabs-flasher

the command you'd want to use is:

`$ universal-silabs-flasher --probe-method bootloader --device socket://IP_ADDRESS:6638  flash --firmware MGM21_2022_ncp-uart-7.2.3.gbl `

davidrule1969, 2023-10-28:
OK, thanks.  That was it.  I was using the wrong .gbl file.  ~ prompt restored and ZHA is now seeing the device! :-)

Now to start playing with experimental zigbee2mqtt :-D

Thanks again for the help.

LordNex, 2023-10-31:
Along those lines will we be seeing a new version soon? I didnt know about the 10/X bug but I havent let ESPHome write to it yet, although the yaml file looks like it just repoints it to your code. Its mostly just annoying as i like to try and keep everything's at the updated stable version,

LordNex, 2023-11-04:
Is it safe to let ESPHome to update the coordinator now that HA11 is out?

mtnbiker717, 2023-11-09:
> For z2m you if you choose to go that way despite the warnings with a efr32 coordinator being experimental. You have to make sure under the port line you tell it ezsp otherwise it assumes a TI adaptor which is why you see the error you got.

What issues are they seeing with Z2M?

closed 2024-03-20 by davidrule1969