tube0013/tube_gateways #164: Error in log while attempting to connect to API

LordNex, 2023-11-02:
The device shows up in my ESPHome, just as adopted, it's still running your firmware. But even though I've added a Custom DNS entry on my internal dns server and pointed it to it, it won't connect. Just shows the below error

'''

WARNING tubeszb-efr32-mgm210-poe-2023.local: Connection error occurred: [Errno 104] Connection reset by peer

'''

tube0013, 2023-11-02:
you could try deleting the esphome device and adding manually from HA ESPHome integration.

LordNex, 2023-11-02:
That’s what I’ve done. I configured the integration separate that ESPHome. All I did was adopt it. Of course it wants to update to 10.5 but I stopped that and make sure only your firmware is loaded. But it shows as online and needing an update so I know it’s talking. But anytime I click logs it gives me a bunch of warnings about the pull up pin then starts trying to connect to its API and that’s when the error starts.

Benjamin T Powers

LordNex, 2023-11-15:
Looks like updating to ESPHome 11.0 corrected this

closed 2023-11-15 by LordNex