tube0013/tube_gateways #167: tubeszb-cc2652-poe-2023 problem ( crashing? )

Bloodpack, 2023-11-18:
Hello i have the same issue as described here https://github.com/tube0013/tube_gateways/issues/154#issue-1866082793 but with the tubeszb-cc2652-poe-2023

i have added recently 6 of these temperature sensors [LYWSD03MMC](https://www.zigbee2mqtt.io/devices/LYWSD03MMC.html#custom%2520devices%2520(diy)-lywsd03mmc) with firmware from [this repo](https://github.com/devbis/z03mmc) the firmware version is [1.0.6](https://github.com/devbis/z03mmc/releases/tag/1.0.6)

since the i get every 26 hours this error in zigbee2mqtt

```
error 2023-11-17 16:41:41: Error while starting zigbee-herdsman
error 2023-11-17 16:41:41: Failed to start zigbee
error 2023-11-17 16:41:41: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-11-17 16:41:41: Exiting...
error 2023-11-17 16:41:41: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

and zigbee2mqtt crashes and it can not be restarted until i manually restart my tubeszb-cc2652-poe-2023 stick from the webui
the webui is responsive but i think the zigbee chip is crashed

after zigbee2mqtt is running again i get this errors every 2 hours in the LOG

```
error 2023-11-17 18:42:58: Adapter disconnected, stopping
error 2023-11-17 18:42:58: Failed to stop Zigbee2MQTT
error 2023-11-17 20:42:58: Adapter disconnected, stopping
error 2023-11-17 20:42:58: Failed to stop Zigbee2MQTT
error 2023-11-17 22:42:58: Adapter disconnected, stopping
error 2023-11-17 22:42:58: Failed to stop Zigbee2MQTT
error 2023-11-18 00:42:58: Adapter disconnected, stopping
error 2023-11-18 00:42:58: Failed to stop Zigbee2MQTT
error 2023-11-18 02:42:58: Adapter disconnected, stopping
error 2023-11-18 02:42:58: Failed to stop Zigbee2MQTT
```

and so on until **_i assume the zigbee chip crashes_** and zigbee2mqtt can not restart because of `Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`

here my zigbee2mqtt versions:

Zigbee2MQTT version - [1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) 
Coordinator type - zStack3x0
Coordinator revision - 20230507
Frontend version - 0.6.142
Zigbee-herdsman-converters version - 15.106.0
Zigbee-herdsman version - 0.21.0

this errors happened first on the tubeszb-cc2652-poe-2023  firmware version `20221226`

i did update the zstack firmware to [20230507](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip) and the esp-home firmware thru the webui to version [051523](https://github.com/tube0013/tube_gateways/raw/main/models/current/tubeszb-cc2652-poe-2023/firmware/esphome/tubeszb-cc2652-poe-2023(051523).bin)

well that did not help same problem......

i think i need some help please

tube0013, 2023-11-18:
Can you try the test fw from her to see if it makes a difference: 

https://github.com/Koenkk/Z-Stack-firmware/discussions/483

tube0013, 2023-11-18:
Also if you send a reset zigbee module from the web front end, is z2m then able to reconnect? If so this is definitely a z-stack fw issue. I have some of those devices on order and plan to flash with the zigbee fw so I will see if they are noisy like some of the tuya devices that seem to be crashing the fw as well.

Bloodpack, 2023-11-18:
> Can you try the test fw from her to see if it makes a difference:
> 
> [Koenkk/Z-Stack-firmware#483](https://github.com/Koenkk/Z-Stack-firmware/discussions/483)

How can i flash having the file downloaded? 

i flashed the 20230507 FW using your addon for homeassistant and adding the link to the firmware...

sorry for the noob question

tube0013, 2023-11-18:
🤦‍♂️ I see the problem. I'll have to post it to my repo and I'll send a link here once I'm on my computer today.

Bloodpack, 2023-11-18:
> 🤦‍♂️ I see the problem. I'll have to post it to my repo and I'll send a link here once I'm on my computer today.

no need i just uploaded the files to my github repo

i will report back.......

Bloodpack, 2023-11-21:
i have tested both firmware versions 20231111 and 20231112 from your link but nothing changed, every 24-26 hours i haver to manually restart my `tubeszb-cc2652-poe-2023` from the webinterface in order to get a functional zigbee2mqtt again.

for now i am running a script to restart my `tubeszb-cc2652-poe-2023` stick every 8 hours and so far it is working well.....

feel free to close here because i think there is nothing you can do, this may be firmware related

tube0013, 2023-11-21:
Instead of restarting have you tried just resetting the zigbee module via the web front end? Whether this works or not to get z2m working again would be helpful to know. 

Thanks

tube0013, 2023-11-21:
Nope, thanks for testing!

Bloodpack, 2023-11-22:
i have tried the `Zigbee Module Reset` option in the web front end of the coordinator and it does not work, getting same errors.....

it starts with this errors in the log

```
debug 2023-11-22 10:03:16: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-11-22 10:03:16: Logging to console and directory: '/config/zigbee2mqtt/log/2023-11-22.10-03-16' filename: log.txt
debug 2023-11-22 10:03:16: Removing old log directory '/config/zigbee2mqtt/log/2023-11-22.08-00-15'
info  2023-11-22 10:03:16: Starting Zigbee2MQTT version 1.33.2 (commit #unknown)
info  2023-11-22 10:03:16: Starting zigbee-herdsman (0.21.0)
debug 2023-11-22 10:03:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[119,73,46,4,0,100,247,144],"networkKey":"HIDDEN","panID":33193},"serialPort":{"adapter":"zstack","path":"tcp://192.168.178.198:6638"}}'
error 2023-11-22 10:03:35: Error while starting zigbee-herdsman
error 2023-11-22 10:03:35: Failed to start zigbee
error 2023-11-22 10:03:35: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-11-22 10:03:35: Exiting...
error 2023-11-22 10:03:35: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```
then i get this 2 times 

```
debug 2023-11-22 10:04:44: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-11-22 10:04:44: Logging to console and directory: '/config/zigbee2mqtt/log/2023-11-22.10-04-44' filename: log.txt
debug 2023-11-22 10:04:44: Removing old log directory '/config/zigbee2mqtt/log/2023-11-22.10-00-16'
info  2023-11-22 10:04:44: Starting Zigbee2MQTT version 1.33.2 (commit #unknown)
info  2023-11-22 10:04:44: Starting zigbee-herdsman (0.21.0)
debug 2023-11-22 10:04:44: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[119,73,46,4,0,100,247,144],"networkKey":"HIDDEN","panID":33193},"serialPort":{"adapter":"zstack","path":"tcp://192.168.178.198:6638"}}'
error 2023-11-22 10:04:44: Error while starting zigbee-herdsman
error 2023-11-22 10:04:44: Failed to start zigbee
error 2023-11-22 10:04:44: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-11-22 10:04:44: Exiting...
error 2023-11-22 10:04:44: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:198:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)

debug 2023-11-22 10:10:46: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-11-22 10:10:46: Logging to console and directory: '/config/zigbee2mqtt/log/2023-11-22.10-10-46' filename: log.txt
debug 2023-11-22 10:10:46: Removing old log directory '/config/zigbee2mqtt/log/2023-11-22.10-00-40'
info  2023-11-22 10:10:46: Starting Zigbee2MQTT version 1.33.2 (commit #unknown)
info  2023-11-22 10:10:46: Starting zigbee-herdsman (0.21.0)
debug 2023-11-22 10:10:46: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[119,73,46,4,0,100,247,144],"networkKey":"HIDDEN","panID":33193},"serialPort":{"adapter":"zstack","path":"tcp://192.168.178.198:6638"}}'
error 2023-11-22 10:10:47: Error while starting zigbee-herdsman
error 2023-11-22 10:10:47: Failed to start zigbee
error 2023-11-22 10:10:47: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-11-22 10:10:47: Exiting...
error 2023-11-22 10:10:47: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:198:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

then this again

```
debug 2023-11-22 10:12:15: Loaded state from file /config/zigbee2mqtt/state.json
info  2023-11-22 10:12:15: Logging to console and directory: '/config/zigbee2mqtt/log/2023-11-22.10-12-15' filename: log.txt
debug 2023-11-22 10:12:15: Removing old log directory '/config/zigbee2mqtt/log/2023-11-22.10-01-04'
[... cut ...]
error 2023-11-22 10:12:34: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

and then when i press `Restart the Device` on the coordinator web front end zigbee2mqtt starts just fine

danhogwood, 2023-11-23:
+1 - I had my tubeszb-cc2652-poe-2023 shipped to me yesterday and set it up and have already experienced a number of crashes as described above. The error messages etc are all the same.

I'll be trialling the auto restart as mentioned here - https://github.com/tube0013/tube_gateways/issues/154#issuecomment-1782190055

Bloodpack, 2023-11-23:
> , thanks for testing!

The restart script works fine for me, no more error messages and zigbee2mqtt runs fine

mrosun, 2023-11-23:
I am having the same exact issue with my PoE coordinator. 

One thing I tested was to revert home assistant to 2023.10 from a backup without restarting the PoE coordinator and it worked, so I suspect there's something going on with the newest versions of HA but I can't tell what it is.

NickN-AH, 2023-11-23:
Same here with the 2022 PoE coordinator. Since updating to home assistant 2023.11, but I'm running with 20220219 firmware.

tube0013, 2023-11-23:
Is everyone running zigber2mqtt? Or are some using ZHA? Home Assistant updates should have no impact on zigbee2mqtt.

NickN-AH, 2023-11-23:
I'm running zigbee2mqtt, but somehow the HA updates affected it. Didn't change anything else.

danhogwood, 2023-11-23:
I am running Zigbee2MQTT, hosted outside of HA so no HA issue/dependency for me

danhogwood, 2023-11-23:
I have created a new issue (but it may be related) here - my system is hitting critical issues every 2 hours - https://github.com/tube0013/tube_gateways/issues/172

bsaurusrex, 2023-11-24:
> > +1 - I had my tubeszb-cc2652-poe-2023 shipped to me yesterday and set it up and have already experienced a number of crashes as described above. The error messages etc are all the same.
> > I'll be trialling the auto restart as mentioned here - [#154 (comment)](https://github.com/tube0013/tube_gateways/issues/154#issuecomment-1782190055)
> 
> The restart script works fine for me, no more error messages and zigbee2mqtt runs fine

I have updated the automation to be event based. Doing this does sometimes leave some devices stranded and requiring a re-pairing.  While that is a bit annoying, the crashing problem isnt noticeable in my day2day.

First, enable the binary_sensor:
binary_sensor.zigbee2mqtt_running
Then you can reference it here:

```
alias: Zigbee2Mqtt_Crash
description: ""
trigger:
  - type: not_running
    platform: device
    device_id: UPDATEME
    entity_id: UPDATEME
    domain: binary_sensor
    for:
      hours: 0
      minutes: 0
      seconds: 10
condition: []
action:
  - service: rest_command.reboot_zb_coordinator
    data: {}
  - delay:
      hours: 0
      minutes: 0
      seconds: 10
      milliseconds: 0
  - service: hassio.addon_start
    data:
      addon: 45df7312_zigbee2mqtt
mode: single
```

tube0013, 2023-11-27:
What network routers are folks using in this issue? Any PFSense? (https://github.com/tube0013/tube_gateways/issues/77)

bsaurusrex, 2023-11-27:
> What network routers are folks using in this issue? Any PFSense? (https://github.com/tube0013/tube_gateways/issues/77)

Ubiquiti

mrosun, 2023-11-28:
> What network routers are folks using in this issue? Any PFSense? (#77)

pfSense. The coordinator is set to get a static ip reservation from the DHCP server. However, the web interface was always working when I checked it, so it seemed to never be an issue. However, I will concede that the new version of pfSesnse just came out recently, in fact, this issue did start happening for me after I installed the newest version, 2.7.1.

LordNex, 2023-11-28:
Of and personally I'm a Fortigate person. I'm certified by them and sell their equipment. But at the end of the day I prefer my Firewalla Gold because it's rock solid and lets me do anything I can dream of without half the headache I already do all day for a living. pFsense can be a bitch if not setup right. Better to go WRT

pugmandan, 2023-11-28:
> > What network routers are folks using in this issue? Any PFSense? (#77)
> 
> pfSense. The coordinator is set to get a static ip reservation from the DHCP server. However, the web interface was always working when I checked it, so it didn't somehow permanently lose its ip address or anything,. However, I will concede that the new version of pfSesnse just came out recently, in fact, this issue did start happening for me after I installed the newest version, 2.7.1.
> 
> HA and the coordinator are on the same subnet.

I am using PFsense and despite making a static reservation within the Pfsense UI, found the issue persisted until I adjusted the ESPHome config to contain the static reservation also. Otherwise, it looks like the zigbee co-ordinator is dropping its lease every 2 hours, asking pfsense for another and then getting assigned the same one. As a result, it doesn't visibly seem to be losing its static mapping

Bloodpack, 2023-11-28:
> What network routers are folks using in this issue? Any PFSense? (#77)

PFSense here, (VLANS Homeassistant and coordinator in the same VLAN)

i adopted the coordinator to ESP-Home and configured a static IP.....
will give tomorrow an update if the the problem is solved

NickN-AH, 2023-11-28:
I'm also using PFsense. How can I adjust the ESPHome config woth the static reservation also?

Bloodpack, 2023-11-28:
> I'm also using PFsense. How can I adjust the ESPHome config woth the static reservation also?

Do it on your own risk, i am not responsible if you break your firmware.  @tube0013 said the new ESP Home versions are not tested by him to work correctly

Adopt your coordinator in ESP-Home

the edit the configuration and add all parameters from here:

https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-cc2652-poe-2023/firmware/esphome/tubeszb-cc2652-poe-2023.yaml

edit this part  with your IP and remove `#` in front of the lines
 ```
#manual_ip:
    #static_ip: 
    #gateway: 
    #subnet: 255.255.255.0 
```

example:

```
  manual_ip:
    static_ip: 192.168.178.198
    gateway: 192.168.178.2
    subnet: 255.255.255.0
    dns1: 192.168.178.2
```

then save and validate the config
when the config is valid install the new firmware with ESP Home

NickN-AH, 2023-11-28:
Ok. Thanks. But I don't have any knowledge about ESP-Home and how to get it running....

tube0013, 2023-11-28:
If you send me the ip/gateway address/subnet and dns if needed I can get a binary compiled for you

NickN-AH, 2023-11-28:
> If you send me the ip/gateway address/subnet and dns if needed I can get a binary compiled for you

That would be great. How can I send you the details?

tube0013, 2023-11-28:
you can just grab or copy the yaml from here into your esphome config folder. 

https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-cc2652-poe-2022/firmware/esphome/tubeszb-cc2652-poe-2022.yaml

and it should show the device in esphome and let you edit the config to uncomment the static IP setting like explained above.

NickN-AH, 2023-11-28:
Great, it worked! Thanks a lot.

LordNex, 2023-11-28:
> you can just grab or copy the yaml from here into your esphome config folder. [...]

Since its bundled with HA you'll probably get a lot of pulls. Should I change mine, I have it blocked in a VLAN with hairpin and mDNS on so I shouldn't be hitting against your repository unless I try and update and open Internet to that vlan which would be rare

LordNex, 2023-11-28:
My adopted adapter is set to

```
substitutions:
  name: tubeszb-efr32-mgm210-poe-2023
  friendly_name: tubezb-efr32-mgm210-poe-2023
packages:
  tubeszb.efr32_mgm210_poe_2023: github://tube0013/tube_gateways/models/current/tubeszb-efr32-MGM210-poe/firmware/esphome/tubeszb-efr32-mgm210-poe-2023.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: Rb3guRvp3+odkfn4TpeG4C3MA7ORUQgE3rsdZWHQZDc=
```

tube0013, 2023-11-28:
That's fine but to specify a static IP you need to edit what is in the original yaml, so you have to pull that

LordNex, 2023-11-28:
> That's fine but to specify a static IP you need to edit what is in the original yaml, so you have to pull that

I have a reservation for it in dhcp and a DNS A record for its mDNS name.  Not mention my Firewalla can handle a lot. I bought the gold but the code is open source. Pretty much any Linux box that has the power can run it. Although their hardware is top notch, with the money, and worth the headache. Made by Cisco engineers who got tired of having to tech at night. Cause thief moms camera got hacked.

LordNex, 2023-11-28:
[image]

LordNex, 2023-11-28:
That's somewhat out of date but you get the general layout without going into SVI structure

tube0013, 2023-11-28:
People have what they have, some reason the esp32 and pfsense don't play nice with the dhcp even when the ip is reserved. I can't dictate what people use for their network at their homes.

LordNex, 2023-11-29:
> I am using PFsense and despite making a static reservation within the Pfsense UI, found the issue persisted until I adjusted the ESPHome config to contain the static reservation also. Otherwise, it [...]

Ok for one thing I think we're having a communications issue because you have network router in addition to zigbee routers which are 2 different things.

LordNex, 2023-11-29:
My original question was how many “routers/repeaters” in the zigbee network each chipset could handle that's based on a Poe/serial return could handle. I know each of my LEADVANCE quirked bulbs can only handle about 6 connections each and when your you ring to each around a 2 acer small ranch then it makes a difference. Thatsvwhy I'm wonderinging if I shouldn't create 2 separate zigbee Zones each with its own radio. One dedicated to security and perimiter defence. Where the other handles more of the inner traffic. I bounce a directed IR system to various mirrors around my property with a IR LUX system at the end. Each one scatters vertically and has an adjacent pickup that basically gives me a 4 walls situation around my house. It picks up some of the neatest photos of birds and bats and the occasional unwanted guest.

Bloodpack, 2023-11-29:
My problem seem to be solved with a static ip in the ESP Home yaml

pugmandan, 2023-11-29:
> My problem seem to be solved with a static ip in the ESP Home yaml

Good news @Bloodpack - glad to hear it

Bloodpack, 2023-11-30:
@tube0013 
thanks for the solution to solve this problem.
now running for almost 48h without any crash

should i close here or leave it open for others to read?

rderensy, 2023-12-09:
> Is everyone running zigber2mqtt? Or are some using ZHA?

@tube0013 I received mine yesterday, I first set it up with ZHA and it crashed, to I tried Z2M and it crashed too.

I am also using pfSense and HA and the dongle are in the same VLAN. I'll try the ESP-home stuff soon.

closed 2024-05-08 by Bloodpack