tube0013/tube_gateways #169: Error During ZHA Setup with EFR32-MGM210-POE

amstoneberger, 2023-11-20:
Hi, I am having major issues trying to get my new EFR32-MGM210-POE added in HA via the ZHA integration. I tried adding the device with no modification to the HA configuration.yaml file with no luck, then tried adding the configuration shown [here](https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-efr32-MGM12-eth_usb). In both cases, I am able to progress through the ZHA setup until I get to the point where I have to select to restore a backup network or establish a new network. This is my first foray into Zigbee, so I select establish a network. Then I get a progress screen saying it's loading the next steps but after about 15-20 seconds it just errors out with a generic "Error" message and no helpful information.

I can see in the device webpage that it is getting connection requests from the HA server, and this is what transpires there:

15:40:39 | [D] | [streamserver:074] | New client connected from 192.168.1.0
15:40:40 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
15:40:42 | [D] | [streamserver:074] | New client connected from 192.168.1.0
15:40:42 | [D] | [streamserver:102] | Client 192.168.1.0 disconnected
15:40:46 | [D] | [streamserver:102] | Client 192.168.1.0 disconnected
15:40:46 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
15:40:49 | [D] | [streamserver:074] | New client connected from 192.168.1.0
15:40:49 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
15:41:06 | [D] | [streamserver:102] | Client 192.168.1.0 disconnected
15:41:06 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
15:41:07 | [D] | [streamserver:074] | New client connected from 192.168.1.0
15:41:07 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
15:41:23 | [D] | [streamserver:102] | Client 192.168.1.0 disconnected
15:41:23 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
15:41:24 | [D] | [streamserver:074] | New client connected from 192.168.1.0
15:41:24 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
15:41:41 | [D] | [streamserver:102] | Client 192.168.1.0 disconnected
15:41:41 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF

I have also attached the HA logs, looks like a lot of Python errors during setup, but I can't be sure if it's a configuration issue on my end or HA just not working properly. At the bottom of the log file I see a bunch of EZSP errors and I'm assuming that's related but no clue on what the fix is.

I am running HA in Docker on a Ubuntu machine and didn't see anywhere in the setup docs about Docker considerations. The HA instance seems to be connecting to the controller initially so I didn't think it was a Docker networking error.

Any help would be greatly appreciated!

[image]
[image]
[image]
[image]

[home-assistant_2023-11-20T22-42-03.736Z.log](https://github.com/tube0013/tube_gateways/files/13420052/home-assistant_2023-11-20T22-42-03.736Z.log)

LordNex, 2023-11-20:
Why are you pointing things at a network address. 192.168.1.0 is not a valid IP address that can be routed through your network first off. Depending on what internet you have and just have a simple flat network, then your routers address is 192.168.1.1 which is where things go when they don’t know where to go. 192.168.1.0 and 192.168.1.255 are both reserved for network broadcast. IMHO it sounds more like a networking issue than a Zigbee

Message ID: ***@***.***>

amstoneberger, 2023-11-20:
> Why are you pointing things at a network address. 192.168.1.0 is not a valid IP address that can be routed through your network first off. Depending on what internet you have and just have a simple [...]

Hi LordNex, my home network is setup as a 192.168.0.1/23 network, and haven't had any issues with any devices spanning the 192.168.0.x - 192.168.1.x ranges. For instance my DHCP pool is set for 192.168.0.11-250 (where my desktop PC resides), and my server is at 192.168.1.0. All of my smart home network devices are in the 192.168.1.x range and I've got blocks reserved for different types of devices.

amstoneberger, 2023-11-20:
I also wanted to provide my docker compose details for HA in case that helps:

  homeassistant:
    container_name: homeassistant
    image: "ghcr.io/home-assistant/home-assistant:stable"
    environment:
      - PUID=1000
      - PGID=1000
    volumes:
      - /home/user/homeassistant:/config
      - /mnt/storage:/mnt/storage
      - /etc/localtime:/etc/localtime:mst
      - /etc/letsencrypt:/etc/letsencrypt
    restart: unless-stopped
    privileged: true
    network_mode: host

LordNex, 2023-11-20:
> Hi LordNex, my home network is setup as a 192.168.0.1/23 network, and haven't had any issues with any devices spanning the 192.168.0.x - 192.168.1.x ranges. For instance my DHCP pool is set for [...]

It makes no sense to supernet your network like that unless you're trying to segment off a section for a different VLAN. Standard CIDR for a Class C network is x.x.x.x/24 such as 192.168.1.0/24. That will give you all 254 available addresses in that subnet. I've never seen anyone set a /23 so I'm betting it's a typo. If that is the case and you're using 255.255.255.0 as a subnet mask you're going to have issues.

tube0013, 2023-11-20:
What version of Home Assistant are you running? There was an issue with the 2023.10 release which would prevent the setup of a network coordinator, as far as I know this was fixed in 2023.11

LordNex, 2023-11-20:
Best price of equipment I own

[My Firewalla Gold](https://firewalla.com/products/firewalla-gold-plus?omnisendContactID=61da22b7e5b2120022ca6398&utm_campaign=campaign%3A+Booster+of%3A+Firewalla+Power+Tips%3A+iOS+Spotlight%2C+Flows+%26+Alarms+Feedback%2C+Tricks+for+Muting+%2865545fa87d09620777237c68%29&utm_medium=email&utm_source=omnisend)

amstoneberger, 2023-11-20:
> What version of Home Assistant are you running? There was an issue with the 2023.10 release which would prevent the setup of a network coordinator, as far as I know this was fixed in 2023.11

I'm on 2023.10.5. Thanks for this info, I will see if I can get it going with the .11 update.

LordNex, 2023-11-20:
> > What version of Home Assistant are you running? There was an issue with the 2023.10 release which would prevent the setup of a network coordinator, as far as I know this was fixed in 2023.11
> 
> 
> 
> I'm on 2023.10.5. Thanks for this info, I will see if I can get it going with the .11 update. 

I installed ESPHome and let it detect and adopt mine. There was a bug in the older versions of ESPHome but that looks like it's been fixed in the latest version. Makes it easy to maintain and update. It's part of the Home Assistant Store or at ESPHome.io if your curious

amstoneberger, 2023-11-20:
I appreciate the input from you two, and happy to report that I am now up and running! I never came across bug/release notes online about issues with the 2023.10 release so very glad you pointed that out!

closed 2023-11-20 by amstoneberger