tube0013/tube_gateways #170: Question: SMA-RP Splitter going to External Antennas

LordNex, 2023-11-21:
I was trying to think of ways to increase the range and reliability of my growing Zigbee network (24 devices ATM). I already found a larger and better antenna that came with it and it seemed to help which got me thinking. Could I split that SMA connection and then run them each to a separate 2.4ghz antenna, one at each end of my house. Have you tried this or do you see any issues doing it. If you do, do you have a specific suggestions. I have a few that I've found pretty cheap on Amazon and I'm making sure they are the SMA-RP so that it has the center post. Thanks again as always

MattWestb, 2023-11-21:
Can being done with one T-Split but then you is losing around 3 dB to and from the antenna (1/2 of TX power and the same 1/2 of the TX is being lost) but if you is doing without T-Split you is getting RF miss match and can losing all signal and getting very strange problems.

LordNex, 2023-11-21:
> Can being done with one T-Split but then you is losing around 3 dB to and from the antenna (1/2 of TX power and the same 1/2 of the TX is being lost) but if you is doing without T-Split you is getting RF miss match and can losing all signal and getting very strange problems.
> 
> 
> 
> By the way one Mesh network is being extended by adding good routers for getting good coverage for the sleeping end device and more power from the coordinator is not helping  the situation only making it worse.

On that makes sense. I've been adding more mains powered light bulbs but my home network is over 70 some devices with about half of that being wifi. Now I have good campus grade Aruba IAP335's in a cluster with adaptive spectrum management and other things that help. So I'm sure I have a lot of 2.4 ghz traffic flying around. Luckily I live in the boondocks so I don't have neighbors to deal with. 

I currently have it plugged into a Cisco Catalyzt 2930E 48 port PoE which but I'm pretty sure it only handles class 1. My primary distribution switch, immediately after my Firewall is an Aruba InstantOn 1930 Class 4 PoE that can handle the more higher powered devices. This is what I needed to makes these supped up WAPs function properly. 15w just wasn't enough to even turn the radios on in them. 

Do you think it would be better to put the dongle on a Class 4 powered PoE switch?  

Another question, what if I just went with a larger, more powerful single SMA antenna and put in in my attic?

Thanks for the help guys!

tube0013, 2023-11-21:
IMO Best bet for expanding range is adding routing devices to the mesh sort of spiraling out from the coordinator.

LordNex, 2023-11-21:
> IMO Best bet for expanding range is adding routing devices to the mesh sort of spiraling out from the coordinator.

So PoE power plays no part in it then?  Doesn't need to be PoE+ or anything?

As you can see that's what I've been trying to do with singled and ozram light bulbs, but I'm limited to where I can put those. 

Could you suggest a repeater that's not a light bulb I could plug in somewhere that would work best?

tube0013, 2023-11-21:
Osram/Slyvania bulbs... ugh.. those can be routing black holes. Smartthings used to have a explicit mention of that  a while back.. I got started years ago with Osram bulbs and never had a good zigbee network until I got rid of them all.

Bulbs can be good extenders, or plugs, not everyone needs a router specific hardware like the device I sell, which is just a CC2652 module flashed with Router FW.

LordNex, 2023-11-21:
I picked those up on Amazon because they were cheap and seemed to work ok. But now that I'm adding more and more I can see what you're saying. 

Any devices you suggest specifically?

tube0013, 2023-11-22:
Yes they are compatible with any coordinator. The esp32 is WiFi and BT only, there are newer variants like the H which is zigbee and I think C which can do zigbee as well. But they are rather new and not much out there. 

The big names in zigbee/thread chips used in devices are Silicon Labs (EFR32 chips), Texas Instruments (CC2652 chips), Nordic (nRF52840), NXP (original aqara devices used these), newer entrants from China are Telink (in a lot of more recent tuya stuff)and Bouffalolab (I’m using some GU10 bulbs with these that are branded ewelink which is related to sonoff). There are probably more I don’t know about as well.

LordNex, 2023-11-22:
I'll have to finish flashing out some of my bulbs and everything and see what happens. It's like my kitchen sink. I have a Singled bulb running the sink light and there's a window right there you can look out while at the sink. Even through the bulb is less than 10 feet from the coordinator, I'm getting red lines( bad RSSI) and the contact sensor I added from the bulb 3 feet from it constantly drops. I don't know why it's that part of the house but that might be a place for some form of additional Mains router

tube0013, 2023-11-22:
Sengled bulbs are not routers. (I also believe they have fw issues too)

LordNex, 2023-11-23:
> Sengled bulbs are not routers. (I also believe they have fw issues too)

The come up saying routers in the visual view. I don't know why they end using the LEDADVANCE quirk as the box is sylvania which I thought was a subsidiary of singled. I could be wrong though with all the different models they throw up on Amazon though. 

[image]

LordNex, 2023-11-24:
Additionally can confirm that ESPHome 2023.11.4 tested and is working properly.

LordNex, 2023-11-28:
Just tested with ESPHome 2023.11.6 and all is well 🔔