tube0013/tube_gateways #173: Zigbee2MQTT doesn't work after upgrading tubezb-cc2652p2-eth coordinator to CC1352P2_CC2652P_launchpad_coordinator_20230507

ValiEne, 2023-11-29:
Hi guys, 
Please help me.
After upgrading the coordinator to CC1352P2_CC2652P_launchpad_coordinator_20230507 fw. the Z2M starts normally bun one of my lights/switches are working. 
What info do I need to provide to solve this issue? 

Thanks,

tube0013, 2023-11-29:
Have you tried re-pairing it?

20230507 has seen a lot of issues - KKoenk has been working with users and TI to try and sort them out.

ValiEne, 2023-11-29:
Thanks @tube0013, 
I've tried with three Ikea switches to force delete them (normal delete outputs some errors) and re-add them - all three worked as expected after this.
The biggest problem is to readd the lights because to make them enter in pairing mode is super hard...