tube0013/tube_gateways #176: tubeszb-cc2652-poe-2023 errors/crashing

aaronoaks, 2023-12-09:
Hello, I have the tubeszb-cc2652p7-poe-2023, which I use with Home Assistant with ZHA, and encounter a reoccurring issue. It can be fixed by restarting the tubezb coordinator and reloading the ZHA integration. After a restart, everything works find, and viewing the coordinator web interface shows issues, just the intial connection with the HA server, and all the zigbee things work fine (where 192.168.10.11 is my HA server):

```
17:46:58 | [D] | [stream_server:081] | New client connected from 192.168.10.11
17:46:58 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
```

But often after several hours I start seeing errors showing in the tubeszb web interface, the same errors streaming down repeating over and over:

```
17:43:26 | [W] | [stream_server:168] | Failed to read from client 192.168.10.11 with error 128!
17:43:26 | [E] | [stream_server:145] | Failed to write to client 192.168.10.11 with error 128!
17:43:26 | [W] | [stream_server:168] | Failed to read from client 192.168.10.11 with error 128!
17:43:26 | [E] | [stream_server:145] | Failed to write to client 192.168.10.11 with error 128!
```

Eventually I can no longer access the tubeszb web interface, and zigbee devices stop responding until I power cycle the coordinator.

tube0013, 2023-12-10:
Hi,

What kind of network router are you using?

aaronoaks, 2023-12-10:
Router is pfSense. HA and the coordinator are on the same subnet, and there shouldn't be any firewall rules between them.

tube0013, 2023-12-10:
this is a common issue with pfSense. To work around it, you'll need to compile a ESPHome fw with static IP.

If you are comfortable with ESPHome the Static IP settings can be uncommented and modified as needed in the yaml from here: https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-P7-poe-2023/firmware/esphome

Written up by a user in this issue:  https://github.com/tube0013/tube_gateways/issues/167#issuecomment-1829594990

For the P7, since it is running an ESP-IDF based firmware there is no Web OTA, compiling in ESPHome and uploading through that will work, otherwise you will need to flash over microUSB (on the side of the board, but slightly removing hte device from the case. If needed I can compile the binary and send if you provide the needed ip gateway and subnet.

aaronoaks, 2023-12-10:
Thanks for the info. I believe I have updated the firmware correctly. It's working for now, hopefully stays that way.

mrosun, 2023-12-12:
> this is a common issue with pfSense. To work around it, you'll need to compile a ESPHome fw with static IP.
> 
> If you are comfortable with ESPHome the Static IP settings can be uncommented and modified as needed in the yaml from here: https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-P7-poe-2023/firmware/esphome
> 
> Written up by a user in this issue: [#167 (comment)](https://github.com/tube0013/tube_gateways/issues/167#issuecomment-1829594990)
> 
> For the P7, since it is running an ESP-IDF based firmware there is no Web OTA, compiling in ESPHome and uploading through that will work, otherwise you will need to flash over microUSB (on the side of the board, but slightly removing hte device from the case. If needed I can compile the binary and send if you provide the needed ip gateway and subnet.

Are you considering adding a static ip assignment setting on the web interface at some point?

aaronoaks, 2024-01-22:
No problems since, solution worked well!

closed 2024-01-22 by aaronoaks