tube0013/tube_gateways #177: Older Tubeszb-cc2652-poe: Serial pin out

samip5, 2023-12-22:
There's the middle connector on the board, but what's the pin-out of it?

I would need to troubleshoot why it won't show up in network.

tube0013, 2023-12-22:
See schematic here: https://github.com/tube0013/tube_gateways/blob/main/models/retired/tubeszb-cc2652-poe/gerber_files/Schematic_Zigbee-eth32-2652P2%20POE%20No%20ISO.pdf

samip5, 2023-12-22:
What's the baud rate when connected though the USB and is that only for connecting to zigbee2mqtt via USB or can I debug though it?

tube0013, 2023-12-22:
115200 and those pins are going only to the zigbee module. The olimex esp32 can be debugged from the microUSB on the side of the module. Just don't use PoE Ethernet when connected to usb as you will likely kill you computer's usb or the module or both.

samip5, 2023-12-22:
> The olimex esp32 can be debugged from the microUSB on the side of the module.

That's the USB I'm talking about, aka MicroUSB next to the RJ45 connector, I tried 115200 but got either gibberish or nothing.

tube0013, 2023-12-22:
That's 115200 too.

If you don't get anything I'd suspect the esp32-Poe is dead.

samip5, 2023-12-22:
Does `��` characters count when doing `screen /dev/ttyUSB0 115200`?

tube0013, 2023-12-22:
Yeah you should get the ESPHome logs. 

Send me an email at store@tubeszb.com and I'll see if we can work something out.

samip5, 2023-12-22:
Alright, sent you an email.

closed 2023-12-23 by samip5