tube0013/tube_gateways #178: Watchdog heartbeat timeout: EzspError('EZSP is not running') EF32 MGM1 ZHA Error

loca5790, 2023-12-26:
Watchdog heartbeat timeout: EzspError('EZSP is not running')

The logs were setup per ZHA documentation and the only thing present is"Watchdog heartbeat timeout: EzspError('EZSP is not running'"

I've restarted HAOS

I've logged into the web interface of the ESP32 and see the following:
<html>
<body>

17:31:20 | [D] | [streamserver:102] | Client 192.168.1.5 disconnected
-- | -- | -- | --
17:31:20 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
17:31:21 | [D] | [streamserver:074] | New client connected from 1
17:31:21 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
17:31:38 | [D] | [streamserver:102] | Client 192.168.1.5 disconnected
17:31:38 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
17:31:39 | [D] | [streamserver:074] | New client connected from 1
17:31:39 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
17:31:55 | [D] | [streamserver:102] | Client 192.168.1.5 disconnected
17:31:55 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
17:31:56 | [D] | [streamserver:074] | New client connected from 1
17:31:56 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
17:32:13 | [D] | [streamserver:102] | Client 192.168.1.5 disconnected
17:32:13 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF

</body>
</html>

I've removed the antenna and covered the device in aluminum foil as well.

loca5790, 2023-12-27:
When I compile the bin and set a static it gives me a web interface I've never seen before.

[image]

tube0013, 2023-12-27:
Which device do you have?

The 2023 efr32 PoE?

Yes definitely use a ESPHome binary with static IP set if using PFSense.

What version of HA? 

Thanks!

loca5790, 2023-12-27:
> Which device do you have?
> 
> The 2023 efr32 PoE?
> 
> Yes definitely use a ESPHome binary with static IP set if using PFSense.
> 
> What version of HA?
> 
> Thanks!

I do have the EFR32 POE "tubeszb-efr32-mgm210-poe-2023"

Current Version is 2023.10.5

ESPHome 2023.10.1

I have a bunch of updates I need to start preforming been working on getting my new network setup.  Starting to roll through HA updates.

tube0013, 2023-12-27:
Does it work for some time and then go into the watchdog loop? 

There have been numerous changes to the watchdog in recent versions, I'd recommend getting updated to a more current ha version.

loca5790, 2023-12-27:
It got stuck before even being able to setup.  I updated Home Assistant to latest version and it has now created a new network succesfully.  Thanks!

closed 2023-12-27 by loca5790

LordNex, 2024-01-10:
> Does it work for some time and then go into the watchdog loop? 
> 
> There have been numerous changes to the watchdog in recent versions, I'd recommend getting updated to a more current ha version. 
> 

- Mine keeps crashing but I'm not sure if it's the ZHA addon or the coordinator it self. Anything I should check to see which is causing this?  As far as I can tell the PoE device is still up and its web API is accessible. But I usually have to restart the PoE coordinator to get it to pick back up its network again. Currently, almost, everything is running. I still have a few window and door contact sensors in hard to reach places I haven't readded yet. But here's a screenshot of my topology. I've tried to build in a spiral around my coordinator. But with most of my end devices being battery Aqara devices, I get mixed results. 

[image]

[image]