tube0013/tube_gateways #182: 2023 POE coordinator not working after changing firmware?

ClintElliotMalcolm, 2024-01-22:
I received the tubeszb-cc2652p7-poe-2023 coordinator recently and for a time things were looking mostly good with my zigbee2mqtt setup. I kept having dropped connections and also occasionally had the `SRSP - SYS - ping after 6000ms` error. I decided yesterday to update the firmware and changing the IEEE address to be the same as my old coordinator. However this made it only get the  `SRSP - SYS - ping after 6000ms` error. I've tried a lot of things and haven't been able to make any progress. 

Things I tried to fix:
* Power cycles on Zigbee2Mqtt and the coordinator
* Updating the ESPHome Firmware
* Changing the ESPHome Firmware to have a static IP
* Using the zstack firmware 20221226, 20230507, 20231111, 20231112
* Changing the IEEE address back to the same as the primary

Basic command used.
```
python3 cc2538-bsl.py -evw -i {IEEE} -p socket://{IP}:6638 CC1352P2_CC2652P_launchpad_coordinator_{date}.hex
```

Tried the tube addon to change the firmware and that hasn't worked either. 

No zigpy commands have been able to work. They error out while trying to skip bootloader?
```
Traceback (most recent call last):
  File "/home/clint/zigbee-stuff/venv/lib/python3.9/site-packages/zigpy_znp/api.py", line 694, in _skip_bootloader
    result = await responses.get()
  File "/usr/lib/python3.9/asyncio/queues.py", line 166, in get
    await getter
asyncio.exceptions.CancelledError
```

I don't see any useful logs in the ESPHome dash even when turing on very verbose.

Anything you could suggest I'd really appreciate.

tube0013, 2024-01-22:
When you reflashed ESPHome was it over OTA through ESPHome or via usb? 

If not usb, please try that - I typically suggest this first when that error surfaces.

ClintElliotMalcolm, 2024-01-22:
I tried USB initially and then OTA once I switched to static ip. I'll try USB again and then see what happens.

ClintElliotMalcolm, 2024-01-22:
Yeah another USB flash didn't change anything

tube0013, 2024-01-22:
if you leave z2m shutdown, and boot up the esp does it show serial connected - ie is something else attempting to use the serial too?

if you were able to flash the serial it means the serial stream is working. 

can you try my zigpy-cli tools addon and see if you can run an energy scan, or do a backup

thanks.

ClintElliotMalcolm, 2024-01-23:
Unfortunately none of that works, except flashing by serial... I'll paste in the config and logs.

Nothing connected to serial
[image]
[image]

zigpy-cli addon energy scan config
[image]

Doesn't work:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: TubesZB Zigpy-CLI Tools
 TubesZB Zigpy-CLI Toolsadd-on
-----------------------------------------------------------
 Add-on version: 0.2.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.1.4
 Home Assistant Supervisor: 2023.12.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service zigpy-cli: starting
[15:56:52] INFO: Directoty Exists
[15:56:55] INFO: Running zigpy - info on socket://192.168.1.251:6638
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 694, in _skip_bootloader
    result = await responses.get()
             ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/zigpy", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/decorators.py", line 45, in new_func
    return f(get_current_context().obj, *args, **kwargs)
[... 103 lines cut ...]
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

ESPHome dash logs:
```
16:00:04 | [D] | [button:010] | 'Trigger Zigbee Module Bootloader' Pressed.
16:00:04 | [D] | [switch:012] | 'Zigbee Module Bootloader Pin' Turning ON.
16:00:04 | [D] | [switch:055] | 'Zigbee Module Bootloader Pin': Sending state ON
16:00:05 | [D] | [switch:012] | 'zRST_gpio' Turning ON.
16:00:05 | [D] | [switch:055] | 'zRST_gpio': Sending state ON
16:00:06 | [D] | [switch:016] | 'zRST_gpio' Turning OFF.
16:00:06 | [D] | [switch:055] | 'zRST_gpio': Sending state OFF
16:00:06 | [D] | [main:555] | Delaying ~10 seconds for cc2652p2 to settle
16:00:14 | [D] | [stream_server:081] | New client connected from 192.168.1.99
16:00:14 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state ON
16:00:17 | [D] | [switch:016] | 'Zigbee Module Bootloader Pin' Turning OFF.
16:00:17 | [D] | [switch:055] | 'Zigbee Module Bootloader Pin': Sending state OFF
16:00:17 | [D] | [main:563] | Please try update with cc2538-bsl tool now
16:00:17 | [D] | [main:566] | cc-bsl usage: cc2538-bsl.py -p socket://ip-of-gw:6638 -evw firmware.hex
16:06:24 | [D] | [stream_server:163] | Client 192.168.1.99 disconnected
16:06:24 | [D] | [binary_sensor:036] | 'TubesZB Serial Connected': Sending state OFF
```

ClintElliotMalcolm, 2024-01-23:
That would make a lot of sense. I'll flash that and get back to you.

ClintElliotMalcolm, 2024-01-23:
Yup that was it. Thanks for replying so promptly. Things seems to be working again.

closed 2024-01-23 by ClintElliotMalcolm

tube0013, 2024-01-23:
Awesome. I'll add some docs to make sure the firmware types are clear. I list the version the units ship with in the little insert included with the package, but want it to be more obvious. Thanks for your patience.