tube0013/tube_gateways #183: PoE Coordinator not responding

aaronoaks, 2024-01-27:
Greetings, I have a tubeszb-cc2652p7-poe-2023 that has been working well, but I found this morning none of my sensors responding. We had some storms last night and some power issues, potentially that is related. It now powers on with a red and orange light, but not the green light, using being PoE and the USB port for power (not at the same time). The green light doesn't come on and the connection light doesn't come on the switch (also tried other known working ports and another switch).

tube0013, 2024-01-27:
Can you reach out to me at store@tubeszb.com and I can get a replacement module out to you.

aaronoaks, 2024-02-02:
Been stable since swapping in the replacement PoE module.