tube0013/tube_gateways #184: Failed to read to client 192.168.2.33 with error 128!

frank2604, 2024-02-05:
Hi Byron.

After one of my coordinators stopped sending commands twice in the last few days and had to be restarted, I had a look at the web UIs of all 5 coordinators.

3 out of 4 tubeszb-cc2652p7-poe-2023 continuously show these entries in the log:

```
Failed to write to client 192.168.2.33 with error 128!
Failed to read to client 192.168.2.33 with error 128!
```

[image]

<br class="Apple-interchange-newline">

One tubeszb-cc2652p7-poe-2023 and the tubeszb-efr32-mgm210-poe-2023 have no log entries.

The cc2652p7 with no log entries seems to have hung up and is no longer transmitting commands as described above. All devices of this coordinator are displayed as offline in Z2M.

The IP address is that of the Ubuntu server on which the Docker containers of HA, Mosquitto and the 5 Zigbee2MQTT are running.

So actually the same as issue #172.

As you know, I have set fixed IP addresses on all coordinators, you had already installed the firmware for this before I sent it.

tube0013, 2024-02-05:
By resetting have you removed power from the coordinator or just restarted the esp32 from the web ui?

I would definitely recommend re flashing over usb the ESPHome FW. 

Let me know how that goes.

frank2604, 2024-02-05:
The first time I removed power. Today I just restarted from the web ui.

OK, re flashing means upload the yaml file you send me via mail over the ESP-Home web ui right? Do I have to adjust anything else than the static_ip suitable for the respective coordinator?

Problem is: In the meantime I run ESP home in a Docker container and it seems that this has no auto detection. How can I make the devices visible in the ESP Home Web UI?

Sorry for the stupid questions. I am not yet familiar with working with ESP Home.

tube0013, 2024-02-05:
I would recommend reflashing a binary over the microUSB port. 

For static ip you can pull down the yaml from GitHub and modify for the ip and compile a binary which can be downloaded and flashed over usb.  I can also provide the binary.

frank2604, 2024-02-05:
Sorry, I have no idea how to compile a binary. Would be great if yo can do this for me.

`gateway: 192.168.2.10`
`subnet: 255.255.255.0 `

`static_ip: 192.168.2.191`
`static_ip: 192.168.2.192`
`static_ip: 192.168.2.193`
`static_ip: 192.168.2.194`

Thanks!

tube0013, 2024-02-05:
Okay I'll get to this tonight.

tube0013, 2024-02-06:
Here are the binaries. I included both the legacy versions that use the ESPHome flasher I document in the documentation here, and also the new factory versions which can be flashed with the ESPHome web flasher: <https://web.esphome.io/>
[Archive.zip](https://github.com/tube0013/tube_gateways/files/14182916/Archive.zip)

The yaml is included as well. Wen using static IPs, you can get the devices to show up in esphome by adding a Yaml for each with it's IP configured this ip will take precedence over the mDNS lookup which is likely faily. 

Again I advise flashing these over usb, this does a full erase which is beneficial sometimes.

frank2604, 2024-02-06:
Thanks a lot! I will try it (via usb). Is there anything to consider? Since I'm doing ESP32 flashing and not CC2652 firmware flashing, I don't need to make a backup, right? The pairing will not be lost?

tube0013, 2024-02-06:
Correct, the zigbee side is not affected at all in this process.

frank2604, 2024-02-07:
I have flashed successfully with web.esphome over USB. The device is running again and there have been no issues for 24 hours so far. I will flash the remaining CC2652 P7s again at the weekend.

At the moment I have four CC2652 P7s and one efr32. I will order two more CC2652 P7 in your store. One as a replacement for the efr32 so that all five are the same (also efr32 not recommended with Z2M). The second as a spare part in case one fails.

Would you be so kind to compile a fifth set of bin with the IP 192.168.2.195?

Again - thank you very much.

tube0013, 2024-02-07:
Good to hear, and thank you. here are the binaries for .195
[Archive 2.zip](https://github.com/tube0013/tube_gateways/files/14200336/Archive.2.zip)

frank2604, 2024-02-07:
With what kind of tool(s) do you compile the binaries out of the yaml file? I am on macOS.

tube0013, 2024-02-07:
I just use the esphome addon on a Test VM I use. I believe it can also be done with the esphome command line but i've not tried. It should work on macOS fine.

I have another HA test instance running in a vm on a Mac Mini with UTM, it's the aarch HAOS image and runs natively on the M1 chip no emulation. It's incredibly fast as well. HA forums have details on installing.

closed 2024-02-13 by frank2604