tube0013/tube_gateways #185: Z-Wave PoE Kit unresponsive

jpnwalters, 2024-02-15:
[image]

I recently purchased the Z-Wave PoE kit and I can't seem to pull down an IP address. I've attached a picture showing the LEDs that are lit. I've confirmed the port is a PoE port, it works, and so does the cable. If I power the device over USB and instead plug into a non-PoE port, I see the same LEDs lit and no address is pulled down. I'm happy to help troubleshoot if you have some thoughts.

tube0013, 2024-02-16:
Hi, Sorry for the trouble.

I've seen this a few times now, and have swapped one recently and when I got it back it worked fine for me, so I am not fully sure why this happens.

What type of PoE Switch and Router are you using from your network?

Have/Can you try a different ethernet cable.

You can also try re-flashing the firmware over microUSB, https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-zw-kit

If you want to try a static ip build to see if that makes a difference the Yaml is at the same link, and can be brought into esphome, modified and complied, or I can compile if you provide the ip/gateway and subnet.

jpnwalters, 2024-02-16:
I went ahead and reflashed with your web flasher binary and it booted right up. For what it's worth, it was connected to a Ubiquiti US-8-60W.

closed 2024-02-16 by jpnwalters

tube0013, 2024-02-16:
Good to hear. Interesting, same switch I'm using for testing, and my network.