tube0013/tube_gateways #186: Issues with Zigbee2mqtt Update and Firmware Upgrade

MiNuSMaiD, 2024-03-08:
Hello,
I just updated my Zigbee2mqtt, and it is not able to connect to the tube_zb_gw_cc2652p2_poe Web Server anymore. I tried to update the firmware CC1352P2_CC2652P_launchpad_coordinator_20230507.hex, but I couldn't get it to update. Here is the error message:

error in zigbee2mqtt
```
Zigbee2MQTT:error 2024-03-07 21:37:29: Failed to start zigbee
Zigbee2MQTT:error 2024-03-07 21:37:29: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-07 21:37:29: Exiting...
Zigbee2MQTT:error 2024-03-07 21:37:29: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-03-07 21:37:32: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-03-07 21:37:32: Logging to console and directory: '/app/data/log/2024-03-07.21-37-32' filename: log.txt
Zigbee2MQTT:debug 2024-03-07 21:37:32: Removing old log directory '/app/data/log/2024-03-07.21-33-53'
Zigbee2MQTT:info  2024-03-07 21:37:32: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
Zigbee2MQTT:info  2024-03-07 21:37:32: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:debug 2024-03-07 21:37:32: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.88.232:6638"}}'
Zigbee2MQTT:error 2024-03-07 21:37:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-07 21:37:51: Failed to start zigbee
Zigbee2MQTT:error 2024-03-07 21:37:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-07 21:37:51: Exiting...
Zigbee2MQTT:error 2024-03-07 21:37:51: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-03-07 21:37:54: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-03-07 21:37:54: Logging to console and directory: '/app/data/log/2024-03-07.21-37-54' filename: log.txt
Zigbee2MQTT:debug 2024-03-07 21:37:54: Removing old log directory '/app/data/log/2024-03-07.21-34-15'
Zigbee2MQTT:info  2024-03-07 21:37:54: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
Zigbee2MQTT:info  2024-03-07 21:37:54: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:debug 2024-03-07 21:37:54: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.88.232:6638"}}'
Zigbee2MQTT:error 2024-03-07 21:38:13: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-07 21:38:13: Failed to start zigbee
Zigbee2MQTT:error 2024-03-07 21:38:13: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-07 21:38:13: Exiting...
Zigbee2MQTT:error 2024-03-07 21:38:13: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-03-07 21:38:16: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-03-07 21:38:16: Logging to console and directory: '/app/data/log/2024-03-07.21-38-16' filename: log.txt
Zigbee2MQTT:debug 2024-03-07 21:38:16: Removing old log directory '/app/data/log/2024-03-07.21-34-36'
Zigbee2MQTT:info  2024-03-07 21:38:16: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
Zigbee2MQTT:info  2024-03-07 21:38:16: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:debug 2024-03-07 21:38:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.88.232:6638"}}'
Zigbee2MQTT:error 2024-03-07 21:38:35: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-03-07 21:38:35: Failed to start zigbee
Zigbee2MQTT:error 2024-03-07 21:38:35: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-03-07 21:38:35: Exiting...
Zigbee2MQTT:error 2024-03-07 21:38:35: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-03-07 21:38:38: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-03-07 21:38:38: Logging to console and directory: '/app/data/log/2024-03-07.21-38-38' filename: log.txt
Zigbee2MQTT:debug 2024-03-07 21:38:38: Removing old log directory '/app/data/log/2024-03-07.21-34-58'
Zigbee2MQTT:info  2024-03-07 21:38:38: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
[... 17 lines cut ...]
Zigbee2MQTT:info  2024-03-07 21:39:31: Starting zigbee-herdsman (0.35.1)
Zigbee2MQTT:debug 2024-03-07 21:39:31: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.88.232:6638"}}'
Zigbee2MQTT:debug 2024-03-07 21:39:39: Saving state to file /app/data/state.json
Zigbee2MQTT:error 2024-03-07 21:39:39: Not connected to MQTT server!
Zigbee2MQTT:error 2024-03-07 21:39:39: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
Zigbee2MQTT:info  2024-03-07 21:39:39: Disconnecting from MQTT server
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/app/lib/mqtt.ts:130:21)
    at Controller.stop (/app/lib/controller.ts:214:9)
    at stop (/app/index.js:111:5)
    at process.handleQuit (/app/index.js:117:9)
Using '/app/data' as data directory
Zigbee2MQTT:debug 2024-03-08 23:59:45: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2024-03-08 23:59:45: Logging to console and directory: '/app/data/log/2024-03-08.23-59-45' filename: log.txt
Zigbee2MQTT:debug 2024-03-08 23:59:45: Removing old log directory '/app/data/log/2024-03-07.21-35-42'
Zigbee2MQTT:info  2024-03-08 23:59:45: Starting Zigbee2MQTT version 1.36.0 (commit #86ed71c)
Zigbee2MQTT:info  2024-03-08 23:59:45: Starting zigbee-herdsman (0.35.1)
[... cut ...] 221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://192.168.88.232:6638"}}'
```
error when i upgrade firmware:
```
C:\Users\MiNuS\Downloads\cc2538-bsl-2.1>cc2538-bsl.py -p socket://192.168.88.232:6638  -evw C:\Users\MiNuS\Downloads\cc2538-bsl-2.1\CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
ERROR: could not open port 'socket://192.168.88.232:6638': OSError(22, 'La syntaxe du nom de fichier, de répertoire ou de volume est incorrecte.', None, 123)
```
thanks

tube0013, 2024-03-08:
Is the web front end accessible?

MiNuSMaiD, 2024-03-09:
> Is the web front end accessible?

Yes it is

tube0013, 2024-03-10:
Please see here: <https://github.com/tube0013/tube_gateways/blob/main/Troubleshooting.md#troubleshooting>

You will want to grab the esphome fw for your specific device. do you have the round PoE model or a rectangular?

MiNuSMaiD, 2024-03-11:
I have the round model.

tube0013, 2024-03-11:
Thanks just wanted to make sure I referred you to the right firmware. 

https://github.com/tube0013/tube_gateways/tree/main/models/retired/tubeszb-cc2652-poe/firmware/esphome

The firmware is at the link above for this model. At this point I'd recommend using the idf version, and attempting to flash over USB.

MiNuSMaiD, 2024-03-25:
Hello, I just reinstalled the ESPHome firmware via USB. Now I can access the module with Zigbee2mqtt, so that's great! 
[image]
My question is, should I update the firmware for the C2652p based Gateways using the CC1352P2_CC2652P_launchpad_*.zip file? I tried to do it but it doesn't seem to work. See the screenshot attached for details.
[image]
thanks

MiNuSMaiD, 2024-03-25:
[image]

tube0013, 2024-03-25:
Make sure zigbee2mqtt is stopped before attempting to upgrade fw. Also per your screenshot on the other issue make sure the intelhex dependency is installed. I also have an addon that can update firmware.

https://github.com/tube0013/tubeszb_addons/blob/main/tzb-cc2652-flasher/DOCS.md