tube0013/tube_gateways #190: Instructions on migration from USB Z-Wave stock to TubeZB?

bifurcated, 2024-05-09:
I'm about to embark on a switch from a USB-connected ZWave stick to the TubeZB ZWave POE module.  Is there a way to migrate config, or do I have to exclude/include all devices manually?

tube0013, 2024-05-09:
Check out this answer from the zwave-js developer:

https://github.com/zwave-js/node-zwave-js/discussions/2927#discussioncomment-5227864

skitchbeatz, 2024-09-19:
> I'm about to embark on a switch from a USB-connected Zooz 700 series ZST10 ZWave stick to the TubeZB ZWave POE module. Is there a way to migrate config, or do I have to exclude/include all devices manually?

Did you figure this out? It so, what steps did you take, @bifurcated ? I'm in the same scenario as you. I'd like to restore a backup to the zwave poe module but am unsure how to do that in zwave js.