tube0013/tube_gateways #192: Can't connect to CC2652 P7 Based Zigbee to PoE Coordinator 2023 via BSL

wraith1385, 2024-06-14:
<head></head><span style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; word-break: break-word;"><span style="word-break: break-word;">I'm using Z2M and I need to migrate from one of your older noon-poe coordinators, so I'm working on changing the IEEE address, but I'm running into some issues.  I tried the bsl tool and after clicking the "Trigger Zigbee Module Bootloader" button, I try to run the script and I get the below:<br><br>python3 cc2538-bsl.py -evw --ieee-address 00:12:4b:00:25:90:62:73 -p<span class="Apple-converted-space"> </span><a href="socket://192.168.5.122:6638">socket://192.168.5.122:6638</a>CC1352P2_CC2652P_launchpad_coordinator_20230507.hex<br>ERROR: [Errno 2] could not open port<span class="Apple-converted-space"> </span><a href="socket://192.168.5.122:6638:">socket://192.168.5.122:6638:</a><span class="Apple-converted-space"> </span>[Errno 2] No such file or directory: '<a href="socket://192.168.5.122:6638">socket://192.168.5.122:6638</a>'<br><br>I also tried with the Zigstar tool and after putting it in BSL mode I see "Error while trying to enable BSL mode - check IP address" in the zigstar tool and the below messages show up in the log on the coordinator webpage:</span>
15:22:12 | [D] | [esp-idf:000] | [0;33mW (1408770) httpd_uri: httpd_uri: uri handler execution failed
-- | -- | -- | --
15:22:12 | [D] | [esp-idf:000] | [0;33mW (1408795) httpd_uri: httpd_uri: uri handler execution failed

</span>

mcpowell87, 2024-06-18:
I am doing pretty much the exact same thing (just received my cc2652 P7 coordinator a few days ago), and had the same issue with zigstar.  There were two things I needed to do: 

1. make sure nothing else is hitting that port (I had a zigbee2mqtt instance that was trying to connect to it)
2. disable "AutoBSL" in zigstar

Hopefully that will resolve that particular issue for you.  Unfortunately, I still have had not any luck migrating to that coordinator yet.

wraith1385, 2024-06-20:
Your instructions helped me be able to connect to the coordinator via zigstar, however I have not ben able to get z2m to connect to it when I've tried to switch over.

mcpowell87, 2024-06-20:
Unfortunately that's where I'm stuck as well.  I gave up trying to migrate and was trying to simply connect it to a fresh instance of z2m.  I have even tried all the beta firmwares up to 20240315.

Are you getting the same error message as me?
```
[2024-06-17 23:57:15] error: 	z2m: Error while starting zigbee-herdsman
[2024-06-17 23:57:15] error: 	z2m: Failed to start zigbee
[2024-06-17 23:57:15] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-06-17 23:57:15] error: 	z2m: Exiting...
[2024-06-17 23:57:15] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

wraith1385, 2024-06-20:
Yeah, same for me and I like you tried a fresh Z2M just to make see if it got the same error and it did.

tube0013, 2024-06-20:
Please try reflashing the esp32 side of the coordinator over the micro USB on the Olimex PoE module. To access open case lid and slightly lift the module out of the enclosure until you can access the port. 

Do not connect PoE and USB at same time.

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-P7-poe-2023/firmware/esphome

mcpowell87, 2024-06-20:
I just reflashed the esp32 module using esphome flasher and the firmware you linked, and unfortunately I have the same issue.  I've tried a couple different switch ports and cables and different vlans to rule out any weird network issues as well, but to no avail.  Just for the record, I am using the koenkk/zigbee2mqtt docker image.

wraith1385, 2024-06-24:
I too reflashed the esp32 module and I'm still getting the same error.

wraith1385, 2024-06-27:
@tube0013 any update on this?  Right now this is a completely unusable product that I would very much like to use.

tube0013, 2024-06-27:
Can you confirm if you ever were able to flash the cc2652 module? I'm a bit unclear on that.

I can swap out the Olimex ESP32-PoE - please contact me at store @ tubeszb.com for that.

Thanks.

wraith1385, 2024-06-27:
I was able to with the Zigstar tool. Was never able to with the bsl python script. I'll hit you up on the store.

wraith1385, 2024-07-11:
@mcpowell87 He sent me a replacement unit.  I plugged it in and connected it to my existing Z2M instance and it started working.  I didn't even have to change the ieee address.

mcpowell87, 2024-07-11:
> @mcpowell87 He sent me a replacement unit.  I plugged it in and connected it to my existing Z2M instance and it started working.  I didn't even have to change the ieee address.  

I got my replacement today as well, and this was the exact same experience I had, though I set it up on a separate instance.  I might try to move it to the existing instance, but I'm happy it's working regardless.

tube0013, 2024-07-14:
Having received both coordinators back from @wraith1385 and @mcpowell87 I could confirm that the cc2652 modules were not operating correctly, any communication via network with esp32 tcp stream connection, or on a direct USB connection worked. However I was able to erase and flash z-stack firmware to both coordinators one over USB with the command line flasher from my mac and another with the TubesZB Add-on for flashing the cc2652 radios. Following the successful flash of both devices I was able to form networks, and run energy scans with zigpy-cli showing them working as expected. 

In the end I think the firmware was corrupted somehow.

closed 2024-07-14 by tube0013