tube0013/tube_gateways #193: Problem with OTA update of ESPHome

nikolai2111, 2024-06-20:
After the latest update to ESPhome (2024.6.1) we had a problem with the OTA update.

It says that the `'platform: esphome'` should be added, but it's not clear where and in the .yaml specified [here](https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-cc2652-poe-2023/firmware/esphome/tubeszb-cc2652-poe-2023.yaml) does not contain any of this.

```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/tubeszb-cc2652-poe-2023.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality
```

nikolai2111, 2024-06-20:
Solution:
Add `platform: esphome` under the `ota:` option.

```
ota:
  platform: esphome
```

nikolai2111, 2024-06-20:
Now the following message appeared:
```
11:31:33 | [W] | [api.connection:107] | : Reading failed: BAD_INDICATOR errno=11
```

tube0013, 2024-06-20:
ESPHome 2024.6.x has many changes I don't advise updating at this time until it can be tested fully. You can go back to older version by using the bin file available for your device.

The api error has to do with the connection to home assistant. Power cycle the device and see if that resolves that. 

If the device shows it has an update just disable the update entity in the devices esphome configuration page. 

There really is no reason to update the device for every ESPHome release.