tube0013/tube_gateways #201: I think my EFR32 MGM21 PoE Coordinator died

tfyoung, 2024-08-04:
It worked wonderfully since it arrived at the start of the year, powered by a tplink POE switch, but a few weeks ago it stopped working after a brief power outage of the switch. I've been pretty busy so haven't diagnosed much, but tonight I tried a few things.

When testing POE or just independent USB power with no network, the two power LEDs come on, but no link lights or any other lights.

I was trying to just flash the olimex poe via the arduino ide as a starting point. The computer detects the serial port, but wouldn't flash:

esptool.py v4.6
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed chip erase: uploading error: exit status 2

I got this both with the board as is, with the EFR module unplugged and holding GPIO0 to ground while powering on.

I then pulled out the multimeter. With it plugged in to USB the 5v pin reads 5v, but worryingly the 3.3v pin is 4.7v.

I also tried https://web.esphome.io/ with it failing everything beyond the initial connect to com port step.

It kind of feels like something on the olimex board has failed and fried my esp32?

Do you have any suggestions for what else to look at?

tube0013, 2024-08-04:
Hi, can you reach out to me at store@tubesZB.com with the name and email used for the order? Sounds like the Olimex ESP32-PoE board died.  Unfortunately there have been reports of the similar in the past I've not yet been able to re-produce. I'll ping Olimex again though..

PatrickHuetter, 2024-09-01:
@tube0013 Seems that i'm affected by the same problem. I had no power outage and the gateway was working fine for months. Did no firmware related things. It just stopped working. I'm using also POE connected to a Unifi Switch. I've the following model: CC2652 P7 Based Zigbee to PoE Coordinator 2023

tube0013, 2024-09-01:
I'll ask that you reach out to me at the store email, please let me know the email address used on the order as well.

PatrickHuetter, 2024-09-13:
@tube0013 I’ve sent you an email. :) Thank you very much for your help!