tube0013/tube_gateways #203: Z-Wave Adapter product page shows incorrect assembly

mag1024, 2024-08-09:
Hi -- I just received my order of the "Raspberry Pi GPIO to Olimex UEXT for Z-Wave Adaptors", assembled it, and it doesn't work -- Z-Wave JS cannot connect.
I believe it may be because the image on the product page, which I followed as my assembly guide, is backwards --the pins are inserted from the wrong side.
[image]

tube0013, 2024-08-09:
Apologies for the incorrect photo.

I can send a replacement se pre-soldered if you'd like. Please let me know over email store@tubeszb.com

Thanks.

Dinth, 2024-08-17:
Hi. I think i have just done the same. No problem with resoldering it myself (i hope that the neither esp board nor razberry got damaged), but how exactly should it be soldered?

tube0013, 2024-08-17:
I removed the incorrect picture. The pins are labeled on the board, this cropped photos from the full kit shows the correct orientation:

[image]

If you need a replacement and don't want to de-solder please reach out to store email.

RyanWor, 2024-09-24:
Just wanted to note the product page is still showing the incorrect assembly. Just put one of mine together and wished I had seen this first 😆

[image]
[image]

tube0013, 2024-09-24:
Apologies, I thought I had updated the photo, even had the cropped photo on my phone, but it must not have saved. I did just update as well as added a view of an assembled adapter with zwave module installed to show orientations.