tube0013/tube_gateways #204: Random unresponsivness

mykpoz, 2024-08-26:
I have an issue similar to https://github.com/tube0013/tube_gateways/issues/179

My PoE coordinator will become unavailable and Z2Mqtt will not be able to communicate with it until I restart the device.

It's random, and don't realize that it isn't working until random automation's have failed to run like turning on the espresso machine to let it warm up before we wake up.

If there is something that I can provide to help try to troubleshoot why this is happening, I will provide what information I can.

Thanks in advance, and sorry for the vagueness... I haven't found any relevant data to share

mykpoz, 2024-08-29:
when it happens it is not pingable, so i assume it is locking up for some reason

tube0013, 2024-08-30:
Having some logs would help, one thing you could try would be running over not powered Ethernet and connecting the micro USB port on the board to computer and pull up the console log, and keep it open until this happens, that would help a lot.

you can also try a static IP for your device: https://github.com/tube0013/TubesZB-ESPHome-Builder

mykpoz, 2024-09-17:
let me know if you need anything else.

tube0013, 2024-09-17:
Did you try flashing the Olimex ESP32 ESPHome firmware with a static IP build? this will help ruled out some issues and also give that device a reset to see if it helps improve things. 

The logs really don't give any insight, they just say the socket closed. can you enable debug logging if it continues.