tube0013/tube_gateways #207: zwave poe board.

JR2012thman, 2024-10-01:
This is perhaps a silly question but I wasn't able to connect to my tubez zwave POE board. How do I go about flashing the firmware again? I tried downloading the bin from the github and flashing that via the esphome web interface but It did not come back to life.

tube0013, 2024-10-01:
Can you explain what is happening?  when originally plugged into powered ethernet what LEDs came on? you should see a red and orange for power and green when ethernet connects.

JR2012thman, 2024-10-01:
I have all three lights, but its not available on the ip address or
discoverable in home assistant.

tube0013, 2024-10-01:
are you able to go to the IP in a browser?

JR2012thman, 2024-10-02:
that was why I reflashed the firmware.bin because I couldn't get to the ip
on the port described on your site.

tube0013, 2024-10-02:
they are 2 different things.

The network serial port for connecting zwave-js is tcp://IP:6638

There is a esphome web frontend running at http:/IP